jschuh/klipper-macros #8: cura slicer issue - first_layer_print_min

kylegl, 2022-09-26:
Hello,

Thanks for putting this up! 

Quick question, I am using cura and trying to get this to print. I get this error when using the PRINT_START macro

`Unable to parse parameter 'MESH_MIN'`

Any idea how I can get around this?

Thanks!

jschuh, 2022-09-26:
The slicer config relies on the Prusa Slicer's [gcode variables](https://help.prusa3d.com/article/macros_1775#variables-placeholders) for the current print state, and its support for custom gcode at layer changes. When I looked at Cura I couldn't find [equivalent functionality](https://files.fieldofview.com/cura/Replacement_Patterns.html).

It's possible that someone more familiar with Cura could make a viable config. Cura's plugin system also appears to provide the necessary functionality, so I'm pretty confident it could be done as a plugin. Unfortunately, I don't use Cura, so it's not something I plan to work on.

In case you want to dig into it, here's the list of custom gcode events and corresponding state that the klipper-macros would need (the missing ones for Cura are italicized):

- start gcode
  - first layer extruder temperature
  - first layer bed temperature
  - _first layer bounding rectangle (MESH_MIN/MESH_MAX)_
  - _total layer count_
- end gcode
- _before layer change_
  - _current layer_
  - _current height_
- _after layer change_

jschuh, 2022-09-26:
Actually, now that I look at this more I think I could probably make it work with Cura's Post Processing extension. The Post Processing extension already has all the necessary layer change support, plus it supports regex gcode replacement, and Cura embeds all the necessary state as comments at the beginning of the file. So, I could add a new macro that saves off print start state, use gcode substitutions in Cura Post Processing to call that macro on every line of needed state, and then just pull any missing state from that if it's not passed to PRINT_START explicitly.

I'll take a look at it sometime in the next few days.

jschuh, 2022-09-26:
Cura should work now. Follow the [instructions here](https://github.com/jschuh/klipper-macros#ultimaker-cura).

closed 2022-09-26 by jschuh

kylegl, 2022-09-26:
Great thanks for the info!

Side note: I tried to use it with prusa and im receiving this error:

`'gcode_macro _apply_bed_surface_offset:gcode': UndefinedError: 'dict object' has no attribute 'bed_surfaces'` 

I have not uncommented anything in the **gcode_marcro _km_options** in my **printer.cfg**

I looked at the **bed_surfaces.cfg** and line 26-27 may have an extra quote. I honestly don't know jinja too well so I could be wrong.

```
{% if s|length > km.bed_surface_max_name_length or
          s|list|select("in", " \r\n\"\'")|list %}
```

Thanks again for the cura setup. I will test it out!

jschuh, 2022-09-27:
You need to name at least one bed surface. Just add something like this to `gcode_macro _km_options` if you have only one surface:

```
variable_bed_surfaces: ['smooth_1']
```

jschuh, 2022-09-27:
No, the code's correct. Did you add all the necessary config sections [listed in the README](https://github.com/jschuh/klipper-macros#klipper-setup)? Because that seems like the sort of error you'd get if Klipper doesn't have a properly configured `save_variables` section.

I just ran a very short print job with the following for `_km_options` as a test on one of my printers:

```
[gcode_macro _km_options]
description: Custom configuration options for this printer.
gcode: # This section is required by Klipper, but not used.
```

I verified that it works fine with no custom variables. So, something must not be set up properly in your `printer.cfg`.

kylegl, 2022-09-27:
The issue was I didn't have a **klipper_config** directory. 

I installed this repository in the same directory as my printer.cfg but never added klipper_config. After I made the klipper_config directory (in the same directory as my printer.cfg) I no longer received that error. Thanks for taking the time to help me out. Much appreciated!