jschuh/klipper-macros #10: [PRINT_START] extruder gets to final temperature and start oozing before leveling the bed

HitLuca, 2022-09-27:
I think it would be better to bring the extruder to printing temp after having the bed leveled, since in the current state it would start oozing for a while if the mesh to probe is big.

I suggest going from

```cfg
[gcode_macro print_start]
gcode:
...
    {% if km.start_level_bed_at_temp %}
        M104 S{EXTRUDER} # set the final extruder target temperature
        G28 Z # Re-home only the Z axis now that the bed has stabilized.
        BED_MESH_CALIBRATE{% if MESH_MIN %} MESH_MIN={MESH_MIN}{% endif
                        %}{% if MESH_MAX %} MESH_MAX={MESH_MAX}{% endif %}
        PARK
    {% endif %}
```

to

```cfg
[gcode_macro print_start]
gcode:
...
    {% if km.start_level_bed_at_temp %}
        G28 Z # Re-home only the Z axis now that the bed has stabilized.
        BED_MESH_CALIBRATE{% if MESH_MIN %} MESH_MIN={MESH_MIN}{% endif
                        %}{% if MESH_MAX %} MESH_MAX={MESH_MAX}{% endif %}
        M104 S{EXTRUDER} # set the final extruder target temperature
        PARK
    {% endif %}
```

jschuh, 2022-09-27:
Did you try setting a lower value for `variable_start_extruder_preheat_scale` in your `_km_options`? The default is 0.5, so it preheats the extruder to 50% of the target temp before starting `BED_MESH_CALIBRATE`, but you could go all the way down to 0.0 (no preheating) if you're seeing oozing.

I've personally found 0.5 works well with my printers, so I don't get oozing but the extruder is just about at temp when `BED_MESH_CALIBRATE` completes, but the ideal value is going to vary from printer to printer.

HitLuca, 2022-09-27:
I see what you mean, in my case my extruder goes up to temp too quickly with larger prints and you get oozing earlier. I'll just use a lower preheat scale, thanks!

jschuh, 2022-09-27:
Hm... I suppose I could scale the preheat and target temps based on the size of the mesh that needs to be probed. That way I could get the extruder temp approaching the target just as the probing finishes. Or maybe before the `BED_MESH_CALIBRATE` I should just use a target temp low enough that oozing isn't likely (e.g. 90% of target temp).

I'll think about it a bit and see if I can come up with a good approach that minimizes oozing during bed probes while reducing the amount of delay waiting for the extruder to warm up.

closed 2023-01-19 by jschuh

Grimones, 2023-02-15:
Hey
My apologies for writing in a closed issue. I was about to open an issue but decided to check first if this was already reported.

Have you come up to any solution? Currently i'm just removing this line 
https://github.com/jschuh/klipper-macros/blob/54334f0a1f58b17e036fc8fea21a29eeaace8d23/start_end.cfg#L63
to avoid oozing. My preheat scale is 0.5-0.6 but it heats pretty fast before the bed mesh calibrate finishes. 

If you have anything to share maybe I could try to implement that

jschuh, 2023-02-15:
It's easy enough for me to add a config variable to just not start the extruder heater until all leveling operations are done. I hadn't because I was just under the impression that setting `variable_start_extruder_preheat_scale` to a low enough value was an adequate solution and other automatic strategies weren't very promising.

jschuh, 2023-02-15:
Setting is there as `variable_start_extruder_set_target_before_level`, although the [`PRINT_START`](https://github.com/jschuh/klipper-macros#print_start) documentation is getting rather long.

Grimones, 2023-02-18:
@jschuh wow, that was fast! thank you for adding that option. It was a bit cumbersome to stash/reset changes to sync with master 😅

jschuh, 2023-05-03:
A few people have asked to be able to set a fixed nozzle probing temperature for things like Voron TAP and load cell probes. I'm thinking of adding that and just removing `variable_start_extruder_set_target_before_level`, since it doesn't seem like both are needed.

Any concerns about that?

HitLuca, 2023-05-03:
Sounds good to me!