jschuh/klipper-macros #11: Error when triggering a M600 command

HitLuca, 2022-09-27:
I'm getting this error when my filament runout sensor triggers a M600

```
Error evaluating 'gcode_macro _load_unload:gcode': CommandError: Command not valid during printing.
```

What could be the problem?

HitLuca, 2022-09-27:
Fixed it, looks like I had a too high value for my extruder `min_extrude_temp` value

HitLuca, 2022-09-27:
@jschuh in order to avoid opening a new issue for just a question I'll write here since it's related to M600.
I believe your macro just keeps waiting for the user to change the filament when a M600 is triggered, but it would be much safer to let the hotend cool down in case the trigger happens during the night

jschuh, 2022-09-27:
It should turn off the extruder heater based on your [`idle_timeout`](https://www.klipper3d.org/Config_Reference.html#idle_timeout) config.

The handler [code is here](https://github.com/jschuh/klipper-macros/blob/main/idle.cfg) and this is the minimal `printer.cfg` needed for it (from [the README](https://github.com/jschuh/klipper-macros/blob/main/README.md#klipper-setup)):

```
[idle_timeout]
gcode:
  _KM_IDLE_TIMEOUT
```

I haven't done a manual filament change in a while, but last I checked it worked correctly.

closed 2023-01-19 by jschuh