jschuh/klipper-macros #12: Bed mesh process produces points that cannot be reached (move out of range)

HitLuca, 2022-10-06:
Haven't had issues with mesh bed leveling until today, and it looks like that for large enough prints the mesh leveling doesn't work properly.I have a 300x300 bed and while trying to print a piece covering almost all width of it I got errors when the macro generated the probing points:

`PRINT_START` command:

```
PRINT_START  EXTRUDER=215 BED=60 MESH_MIN=7.99438,99.422 MESH_MAX=292.006,200.578 LAYERS=495
```

generated points as taken from the console:

```
Move out of range: -40.006 83.422 8.000 [0.000]
Move out of range: -40.006 83.422 8.000 [0.000]

35 | (-40.0, 184.6) | (8.0, 200.6)
34 | (16.8, 184.6) | (64.8, 200.6)
33 | (73.6, 184.6) | (121.6, 200.6)
32 | (130.4, 184.6) | (178.4, 200.6)
31 | (187.2, 184.6) | (235.2, 200.6)
30 | (244.0, 184.6) | (292.0, 200.6)
29 | (244.0, 164.3) | (292.0, 180.3)
28 | (187.2, 164.3) | (235.2, 180.3)
27 | (130.4, 164.3) | (178.4, 180.3)
26 | (73.6, 164.3) | (121.6, 180.3)
25 | (16.8, 164.3) | (64.8, 180.3)
24 | (-40.0, 164.3) | (8.0, 180.3)
23 | (-40.0, 144.1) | (8.0, 160.1)
22 | (16.8, 144.1) | (64.8, 160.1)
21 | (73.6, 144.1) | (121.6, 160.1)
20 | (130.4, 144.1) | (178.4, 160.1)
19 | (187.2, 144.1) | (235.2, 160.1)
18 | (244.0, 144.1) | (292.0, 160.1)
17 | (244.0, 123.9) | (292.0, 139.9)
16 | (187.2, 123.9) | (235.2, 139.9)
15 | (130.4, 123.9) | (178.4, 139.9)
14 | (73.6, 123.9) | (121.6, 139.9)
13 | (16.8, 123.9) | (64.8, 139.9)
12 | (-40.0, 123.9) | (8.0, 139.9)
11 | (-40.0, 103.7) | (8.0, 119.7)
10 | (16.8, 103.7) | (64.8, 119.7)
9 | (73.6, 103.7) | (121.6, 119.7)
8 | (130.4, 103.7) | (178.4, 119.7)
7 | (187.2, 103.7) | (235.2, 119.7)
6 | (244.0, 103.7) | (292.0, 119.7)
5 | (244.0, 83.4) | (292.0, 99.4)
4 | (187.2, 83.4) | (235.2, 99.4)
3 | (130.4, 83.4) | (178.4, 99.4)
2 | (73.6, 83.4) | (121.6, 99.4)
1 | (16.8, 83.4) | (64.8, 99.4)
0 | (-40.0, 83.4) | (8.0, 99.4)

bed_mesh: generated points
Index | Tool Adjusted | Probe
```

This error never occurred with smaller prints, so I think it's just an edge case that needs to be figured out.
I tried to set `variable_probe_mesh_padding` to `0.0` to try and mitigate the issue without luck

Lemme know if you need other info

jschuh, 2022-10-06:
Would you mind sharing the `[bed_mesh]` section from your printer.cfg to help me diagnose?

jschuh, 2022-10-07:
It just occurred to me that I think I know what you're hitting, because it's a mistake I've made before when setting up a new printer. Klipper requires that your `mesh_min` and `mesh_max` account for any offsets for your probe ([example here](https://www.klipper3d.org/Bed_Mesh.html#rectangular-beds)). If you don't do that, Klipper will generate an illegal mesh any time you perform a standard full bed probe (which is what happens when you pass a large enough mesh to `PRINT_START`).

So, without seeing your `bed_mesh` and `probe`/`bltouch` configs, my guess is that the X value in your `mesh_min` is less than the `x_offset` configured for your bed probe. You should be able to fix it by setting that value to `x_offset` plus a few millimeters of padding (and do the same for the Y min/max if necessary).

HitLuca, 2022-10-07:
I have this `[include klipper-macros/*.cfg]` which i'm pretty sure doesn't include subfolders. I will test it out when I get back home

jschuh, 2022-10-07:
That's correct. That line will not include subfolders, so without the explicit include directive the correct `BED_MESH_CALIBRATE` will not get called (hence why the folder is labelled "optional").

Like I mentioned above, this behavior is really a bug on my part. I just wrote a patch to split the code so that the optimized mesh levelling works by default, and the optional part is necessary only if you want the `G20` and `BED_MESH_CALIBRATE` override. I'm testing it now, and if it's good I'll post it in a bit.

It also seems like you weren't actually getting the optimized mesh leveling. It was reducing the mesh size, but not the number of points. So once this is fixed your mesh should run faster for smaller objects.

closed 2022-10-07 by jschuh

HitLuca, 2022-10-07:
I can confirm that adding the missing import fixed my bed leveling issues