jschuh/klipper-macros #13: Problem with variable_bed_surfaces

appleimperio, 2022-10-21:
I recently add a klicky probe now every time I enable the variable_bed_surfaces. I'm getting this error: Error evaluating 'gcode_macro set_surface_active:gcode': TypeError: a float is required

```
[Variables]
bed_surfaces = {'active': u'smooth', 'available': {u'smooth': {'offset': '0'}, u'texture': {'offset': 0.0}}}
```

appleimperio, 2022-10-21:
I found that one of the klicky probe macros was resetting the gcode offset to 0 which i think is the command you use to set the bed surface offset. The strange thing it is saving the gcode state before set it to 0 but when it restored it  i get the error it completely erase the values on the variable file. For now I just comment that line and its working (hoping it don't crash when its doing the docking) if you have a better solution let me know thanks.
```
[gcode_macro _entry_point]
gcode:
    {% set function  = 'pre_' ~ params.FUNCTION %}
    {% set move_accel = printer["gcode_macro _User_Variables"].move_accel|default(1000) %}
    # mandatory to save the new safe position
    M400
    SAVE_GCODE_STATE NAME={function}
    # removes the Z offset for better bed based docking
    #SET_GCODE_OFFSET Z=0
    # all the macros initially assume absolute positioning
    G90
    # set a safe(sane) Acceleration
    SET_VELOCITY_LIMIT ACCEL={move_accel}
```

jschuh, 2022-10-22:
Thanks for the report. It looks like I need to make my SET_GCODE_OFFSET wrapper ensure it's converting to the correct data type prior to saving.

That stated, I don't think that klicky macro should be blindly the clearing the gcode offset. I get what they're trying to do, but the right approach would be to perform the docking operation accounting for the gcode offset.  Or, I suppose save, the offset and restore it when done (although that could still clear it if something aborts the operation at the wrong time).

Is there a github for the klicky probe? I might see about writing a PR.

closed 2022-10-22 by jschuh

appleimperio, 2022-10-22:
Thank You! here is the github for the klicky probe https://github.com/jlas1/Klicky-Probe

jschuh, 2022-10-22:
I thought about it a bit more and I think klippy is doing the right thing here. So, I added code to track when `SET_GCODE_OFFSET` is used in an ephemeral context (i.e. when `SAVE_GCODE_STATE` has been called without a matching call to `RESTORE_GCODE_STATE`). In that case the macro simply doesn't persist the offset change into the saved variables.

You should be able to uncomment that line now, and everything should work as expected.