jschuh/klipper-macros #14: note about relative_reference_index

dtk1985, 2022-10-24:
no issue, just a note for the warning section:

let's say you use 5x5 mesh and enable relative_reference_index in the [bed_mesh] section of printer.cfg and set it to 12 so it references the middle of the bed (##FOLLOW THE FORUMULA FOR relative_reference_index, for example: [(5x5)-1] / 2 = 12), when you want to print something it starts the heatup process, printhead goes to the safe_z_home position and after probing the point klipper throws 3 internal errors for the macros and shuts down. I guess because I was trying to print a small object and your macro switched to 3x3 mesh mode, but still wanted to use the "relative_reference_index: 12" (12th point - non-existent in this case) and the macros could not handle that. I could not figure out for a day what is happening, reinstalled the whole OS lol

jschuh, 2022-10-25:
Sorry, for the hassle. I've never used that and wasn't even aware it existed. I can think of a few different ways to handle this, but the safest and easiest is probably to just disable the mesh size optimization if a relative_reference_index is set.

dtk1985, 2022-10-25:
No worries, actually I turned it on during a print and I forgot about it and after a restart it started throwing the errors. It's off now, don't really need it I realized.

closed 2022-10-25 by dtk1985

jschuh, 2022-10-26:
I landed 650e0ea to disable the fast mesh stuff when `relative_reference_index` is present. I figure it's better than getting bit by some confusing failure.