jschuh/klipper-macros #15: LCD menu Offset tune

appleimperio, 2022-10-26:
Looks like when you fix the G-Code offset for the klicky probe breaks something in the LCD Menu. Every time I want to adjust the offset from the LCD the offset reset by -5 in my case. Thanks

jschuh, 2022-10-27:
Sorry about that. The macros should persist the offset so long as there's currently no saved gcode state in effect, but maybe there's a macro leaving save state hanging open. Would it be possible to run the following command in the console the next time this happens, and paste the output back here?

```
LISTVARS S=state_set
```

appleimperio, 2022-10-27:
As soon as I click on the Tune/Offset Menu on the LCD it jumps to -5 and don't let me go lower if I adjust the offset through the Mainsail interface I have no problem it even saved in the variables file.

[image]

jschuh, 2022-10-27:
Sorry, but I just don't know what's going on here. The recent change for klicky simply prevented the Z offset from being persisted as a saved variable if any saved gcode state was currently active. It doesn't modify the current offset, and the output of `LISTVARS` shows that there's no active saved gcode state anyway when you're seeing this issue.

Maybe you're hitting a conflict with other macros you're running? I try to avoid that with namespacing, but it's certainly possibility. I'll add that I do use the LCD to adjust Z offset on my printers pretty regularly (literally today), so the fact that it's working in my setups certainly gives me the impression it's a conflict with something in your setup.

appleimperio, 2022-10-27:
Ok Let me see what I found. Should I look in the Macros related to the LCD an see if there's another LCD Macro running around causing the conflict?

jschuh, 2022-10-27:
Yeah, I'd look at the gcode for that menu, and for anything else overriding `SET_GCODE_OFFSET`. Unfortunately, weird code conflicts are the unavoidable hazard of monkey-patching.

appleimperio, 2022-10-27:
Got it!!! Thank you, was in the lcd_menus.cfg. When I changed to the kicky probe the offset is lower than -5 and that's the lowest value you setup in that file, increase it to -8 and it's working. Thank you very much!!!

closed 2022-10-27 by appleimperio

jschuh, 2022-10-28:
Hey, I just looked at this and in retrospect I don't think I should change the LCD menus, since it seems like the appropriate thing here is for the user to just update the corresponding `z_offset` when changing out a probe/bltouch.

That stated, it is a pain to have to manually update all your sheet settings when that happens (and it's something I've had to do more than once). So, I was thinking maybe the values for the `z_offset` and `position_endstop` should be tracked by the macros and there should be a simple command to just update all the sheet offsets. I could have a check run at startup that lists when these change, and offers the appropriate command to either update the offsets or ignore the change.

What do you think?

appleimperio, 2022-10-28:
That's even better.