jschuh/klipper-macros #17: Bed heating wait duration

kylegl, 2022-11-14:
Hello,

This is not an issue with your macros. Everything's been working great. I'm trying to turn off the delay between the bed reaching temperature and the print starting. 

I tried setting **variable_start_bed_heat_delay: 0** but it still waits for an amount of time after the bed has reached temp to start the print.

How can I turn this feature off for some prints?

Thanks!

jschuh, 2022-11-15:
The extra delay is probably because the macro overshoots the target temp by a few degrees and then waits for the bed to settle down. I find this useful on my printers (and tune the value for each), but you can shut it off by setting `variable_start_bed_heat_overshoot` to `0.0`.

kylegl, 2022-11-15:
I find it useful as well, but for test or calibration prints I want to make it quick. Ok I'll give that a go. 

How do you tune it?

jschuh, 2022-11-15:
The tuning I do is just by hand. I got the baseline values on a few different beds by taking a reading of the top surface with an IR thermometer and determining how long the top surface took to get up to full temp after the thermistor read full temp. Although, at this point I mostly just estimate based on the mass of the bed, since that works well enough.

Thin PCB heaters (typically found on bed slingers) have relatively small mass, so it's probably fine to significantly lower or even remove the overshoot and delay. The thick MIC 6 beds on a printer like a Voron are a different story, where the surface of the bed can take quite a bit longer to reach the same temperature as that measured at the thermistor. I've also found it helpful on larger PCB beds (e.g. 310mmx310mm) where I would see deviation in the bed mesh if I didn't let it heat up and settle first. Although, in those cases I still don't need anywhere as much delay as I do for a MIC 6 bed.

One more tidbit is that I added the overshoot part to speed things up. During my initial measurements with the IR thermometer it would take quite a bit longer for the top surface to reach the proper temp, but the overshoot cut the delay down significantly. That stated, I haven't really tuned that value on my printers. I think I just keep it at 2.0 for large/thick beds, and 0.0 for thin small and thin beds.

Hope that context helps.

kylegl, 2022-11-17:
Ok, that clears it up for me. I have a glass bed (220 x 220) I set it at .5 overshoot. I don't have an ir thermometer but it seems to be good enough. 

Thanks for the info!

closed 2022-11-17 by kylegl