jschuh/klipper-macros #18: Nozzle temperature doesn't return to original value after cooling down waiting for filament replacement

HitLuca, 2022-11-17:
Happened to me tonight: my printer stopped as the filament ran out mid print, and I only noticed it this morning. The printer sat idle long enough to enter cooldown, so the bed stayed at temperature while the nozzle stopped heating. I loaded the filament using the provided macro and resumed the print, and I noticed that the nozzle temp stayed at the filament loading temperature, and not the printing temperature.
Happy to provide more details if needed

jschuh, 2022-11-17:
This is mostly intentional because I want to let the user adjust the filament temperature at the time of change. That stated, if the printer is currently paused I could have the filament load macro default to heating back to the saved temperature at the time the print was paused if the temperature hasn't been otherwise adjusted. It would be an easy enough change, but I want to think a bit on whether or not it makes sense.

HitLuca, 2022-11-17:
To me if a print is resumed it should return to its previous state, together with the used temperatures. The user can then update them but it should be optional

jschuh, 2022-11-17:
Yeah, I think it probably makes sense. This is the entirety of the change. I just need to make some time to verify it on a test print.

```
diff --git a/filament.cfg b/filament.cfg
index 0fea97d..7e2dd48 100644
--- a/filament.cfg
+++ b/filament.cfg
@@ -10,8 +10,16 @@ gcode:
   SET_GCODE_VARIABLE MACRO=_load_unload VARIABLE=is_printing VALUE="{1}"
   {% set saved_extruder = printer.toolhead.extruder %}
   {% set EXTRUDER = params.EXTRUDER|default(saved_extruder)|lower %}
-  {% set MINIMUM = params.MINIMUM|default(
-           printer.configfile.settings[EXTRUDER].min_extrude_temp + 5.0) %}
+  {% if 'MINIMUM' in params %}
+    {% set MINIMUM = params.MINIMUM %}
+  # This is the special case for a filament change after an idle timeout.
+  {% elif printer.pause_resume.is_paused and printer[EXTRUDER].target == 0 and
+          printer["gcode_macro resume"].saved_extruder_temp %}
+    {% set MINIMUM = printer["gcode_macro resume"].saved_extruder_temp %}
+  {% else %}
+    {% set MINIMUM =
+             printer.configfile.settings[EXTRUDER].min_extrude_temp + 5.0 %}
+  {% endif %}
   {% if MINIMUM < printer.configfile.settings[EXTRUDER].min_extrude_temp %}
     { action_raise_error("Extrude below minimum temp.") }
   {% elif printer.pause_resume.is_paused %}
```

jschuh, 2022-11-17:
Okay, ca5abb1 should make it so the default behavior for an idle timeout during a filament change for a paused print is to reheat the nozzle to the previous temperature unless the load command explicitly specifies a new temperature or the end user has already adjusted the temperature. It's a bit complicated to describe the scenario, but this behavior at least seems most intuitive.

closed 2022-11-17 by jschuh

HitLuca, 2022-11-25:
I just had a filament runout where the Nozzle stayed idle enough to cool down and can confirm the filament loading temperature went back to the previous one when running the filament load macro