jschuh/klipper-macros #19: bed mesh calibration error

solo2424, 2022-11-20:
When trying to print I get the following error. Any idea why?

gcode_macro bed_mesh_calibrate_fast:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro bed_mesh_calibrate'

nodlek-ctrl, 2022-11-21:
Same error for me.

jschuh, 2022-11-21:
That sounds like it's triggering the `BED_MESH_CALIBRATE_FAST` macro when you don't have a `BED_MESH` section specified in your config. There should be guards that prevent that, but maybe I accidentally moved something outside the guards.

Do you have a `BED_MESH` section in your `printer.cfg` file?

solo2424, 2022-11-21:
I moved all my macros into a separate config to keep my printer.cfg
cleaner. Could that be the reason why? I am not home right now but I will
check once I get home.

jschuh, 2022-11-21:
I don't think so. Peeking at the relevant Klipper code, it looks like it reads in and parses the main config and its recursive set of included configs before any modules get a chance to interpret any of the statements. Given that, I wouldn't expect order or location to matter (so long as the required files, paths, and sections are all correct).

It does look like include paths are resolved [relative to the file containing the `include` statement](https://github.com/Klipper3d/klipper/blob/c51f169c06921152a2e9c386016660d0bb09f767/klippy/configfile.py#L217). So, moving things to different directories could result in unexpected behavior if you're not accounting for that.

solo2424, 2022-11-21:
Tahat did it. I uncomment this line([include
klipper-macros/optional/bed_mesh.cfg])  in your config and it is now
working. Not sure why I don't have a bed_mesh config by default. I have
played around so much with my configs that I could have pretty much
accidentally removed it. I am still new to this world so I just do a lot of
trialing and error. Thanks for pushing me in the right direction!

jschuh, 2022-11-21:
Yeah, that's an optional include that's commented out by default because it will throw a macro error if you don't have a `bed_mesh` configured (and a corresponding `probe`, `bltouch`, or whatever to generate the mesh).

Glad it's working for you now.

closed 2022-11-21 by jschuh

chrisleck, 2023-01-15:
I'm just trying these macros for the first time and I'm running into this.  I can confirm my printer.cfg has a `[bed_mesh]` section defined.

chrisleck, 2023-01-16:
It looks like the optional bed_mesh config isn't really optional - the non-optional script somehow assumes it's been called.