jschuh/klipper-macros #26: BUG: Malformed commmand _gcode_wait_wrapper

apphead, 2022-12-18:
Issueing a M109 T1 S205 or M109 T1 S205 (etc)

fails with error
Malformed command '_GCODE_WAIT_WRAPPER HEATER={'pressure_advance': 0.0, 'target': 0.0, 'power': 0.0, 'can_extrude': True, 'smooth_time': 0.04, 'temperature': 16.18} S="20" M="109" T="1"'

jschuh, 2022-12-18:
Wow. That typo has been there since I last had a multi-extruder config up and running. Thanks for the report, and I think it should be fixed now (although I currently don't have a multi-extruder config test it against and won't until next month at the earliest).

apphead, 2022-12-19:
Thanks for the quick fix. Unfortunately, the same error occurs.

I suspect the wrapper code is malformed.

And by the way: the bug you fixed is still there in m104.

jschuh, 2022-12-19:
Ugh, this is what I get for leaving my only multi-extruder printer disassembled for the better part of a year. Anyway, I'll land the fix for `M104` today and take a closer look at the rest of the macros in the file to see if there are any similar issues. And thanks again for the reports.

jschuh, 2022-12-19:
Took a closer look and the extruder selection was more broken than I'd thought at first glance. Looks like I did it when I was merging macros from my different printers prior to the initial Github check-in. It should be fixed now (at least, it seemed to work correctly when I tested against a simulated config).

I've been slowly building an idex from my old dual extruder machine, so hopefully I get that done soon and it will keep me from breaking this in the future. Either way, I certainly would appreciate you filing bugs for any other multi-extruder issues you encounter.

closed 2022-12-19 by jschuh

apphead, 2022-12-20:
> this is what I get for leaving my only multi-extruder printer disassembled
Haha, it took me literally years to complete my homebrewed IDEX, mostly because it was a pain to calibrate the offsets and because I use a not-so-common controller (RADDS). Have tried Marlin, MK4duo, Repetier and now Klipper. Clearly THE best option out there. While still a work in progress, I have made progress over the last few weeks and implemented my own version of https://github.com/HaythamB/TAMV/blob/master/drivers/MoonrakerAPI.py TAMV.py to calibrate offsets. Now waiting for my microscope to arrive in the hope that manual calibration will be a thing of the past.

I have studied your macros, and as far as I can tell, parking does not support multi-tool setup & toolchanging is missing altogether. Correct?
My next step is to test https://github.com/Laker87/klipper integration of duplication / mirror mode, see what changes in my macros / setup it takes and then merge all, including your collection of macros. Long way to go.
Anyway, will keep you posted on bugs etc. and if you are interested, will try to create my own repo with all the code in one place. Happy tinkering!

jschuh, 2022-12-20:
Have to admit I haven't really thought much about calibrating the idex yet. And yeah, the parking macros would have to change, along with some menu stuff, the recently added purging code, etc. I do have a sense of how I plan to approach it (without complicating single extruder configs). I was also planning on making it more tool-changing compliant rather than building tool-changing into the macros, but that view may evolve when I really get into it.

I was also planning on sticking with mainline Klipper, since I expect that the remaining bits of first class idex support aren't too far off. Kevin's feedback on various PRs and some of the online discussions paint a good picture of how he wants it implemented. So, if it's not done by the time I need it my plan was to finish off whatever I need and submit PRs (I've had good experiences contributing there in the past).