jschuh/klipper-macros #27: Error on 'SET_PRINT_STATS_INFO CURRENT_LAYER="-5"': CURRENT_LAYER must have minimum of 0

remigod86, 2022-12-19:
Hello,
After the update from last week (and continued with the newest update), I am getting the following error as soon as the print starts (after the purge line):

Error on 'SET_PRINT_STATS_INFO CURRENT_LAYER="-5"': CURRENT_LAYER must have minimum of 0

followed by

SDCARD_RESET_FILE cannot be run from the sdcard

jschuh, 2022-12-19:
My best guess is that this is a slicer config issue, or maybe a conflict with some other macros you're using. To explain, these macros should never produce a negative layer value (the first error) or call `SDCARD_RESET_FILE` from the sdcard context (the second error).

Could you share the gcode file that's causing the issue? At a minimum it would be helpful to know the full gcode lines for any `PRINT_START`, `PRINT_START_SET`, and `INIT_LAYER_GCODE` macros (you may have just a `PRINT_START` if you're using Prusa Slicer). The other important thing to know is if any `BEFORE_LAYER_CHANGE` statements in that file pass a negative `LAYER` value.

Assuming that the gcode file doesn't contain any statements passing negative layer values, the next most likely culprit is other macros. Last month in 3f1bf6170b12227b5bc8b19bb3862ca3ee31ba44 I switched the layer tracking over to Klipper's newly added `SET_PRINT_STATS_INFO` command (to better integrate with Moonraker and frontends). It's possible you have some other macros that are conflicting in their use of `SET_PRINT_STATS_INFO` (although I don't have a good idea for how that would end up such an unusual layer value).

Bertoman87, 2022-12-19:
Hello! I have the same issue but My is new installed and im not know what im doing 😀. 
First i got SET_PRINT_STATS_INFO
Then this error comes up. 

Error evaluating 'gcode_macro _km_layer_run:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'info'
 i dont know what To do 🙈

remigod86, 2022-12-19:
After you've mentioned the GCODE, I looked inside Cura and realised that "Absolute Extruder Prime Position" was disabled. In my case, it has to be enabled. After enabling it, the first test-print worked flowlesly. Thanks for the help and the great Macros!

jschuh, 2022-12-20:
@remigod86 Glad it's sorted.

@Bertoman87 That doesn't seem like a related problem, but I would guess that the `[virtual_sdcard]` section is missing from your config or you're running an older version of Klipper that doesn't support `SET_PRINT_STATS_INFO`. Please try updating your Klipper install and make sure your config has all the [required sections](https://github.com/jschuh/klipper-macros#klipper-setup).

closed 2023-01-19 by jschuh

TVsIan, 2023-02-03:
> After you've mentioned the GCODE, I looked inside Cura and realised that "Absolute Extruder Prime Position" was disabled. In my case, it has to be enabled. After enabling it, the first test-print worked flowlesly. Thanks for the help and the great Macros!

I'm having this same CURRENT_LAYER issue with Cura (works fine in PrusaSlicer), but I don't seem to have that Absolute Extruder Prime Position available for my Ender 3 S1.

jschuh, 2023-02-03:
Sorry, I worked out a basic config for Cura really just as a favor to a friend. But I don't use it personally, and don't know how to diagnose Cura settings conflicts. Pretty much everything I know is captured in the [Cura setup instructions](https://github.com/jschuh/klipper-macros#ultimaker-cura).