jschuh/klipper-macros #28: When the filament runs out, the M600 command is not executed

raptoreumonline, 2023-01-05:
I would like some advice, I use your macros and
when executing the M600 command, when the filament sensor reports that the filament is running out, it reports this error...

**!! Error evaluating 'gcode_macro _load_unload:gcode': CommandError: Command not valid during printing.**

i use this filament sensor setup:

**[filament_switch_sensor filament_sensor]
pause_on_runout: True
switch_pin: PG12
runout_gcode:  M600**

jschuh, 2023-01-05:
That's very odd, since it works during slicer inserted filament changes. But Klipper's print state heuristic can be a bit fuzzy, and I have a one-line change that should fix the error. I'll try to check it in sometime today.

Regarding my own filament runout sensors, I keep them far enough from the extruder that I can pause at the layer change and ensure I don't mar the surface of the print. The code looks like this:

```
[filament_switch_sensor runout]
pause_on_runout: False
switch_pin: ^PC15
runout_gcode:
  {% if printer.print_stats.info.current_layer <
          printer.print_stats.info.total_layer and
        printer.idle_timeout.state|string == "Printing" and
        not printer.pause_resume.is_paused %}
    PAUSE_NEXT_LAYER
  {% endif %}
```

jschuh, 2023-01-05:
On second thought I just landed the change. Normally I'd test first, but this is a trivial fix.

The unload inside the `M600` should now work without throwing an error about the print state.

closed 2023-01-05 by jschuh