jschuh/klipper-macros #30: Upcoming klipper changes for bed mesh

HitLuca, 2023-01-13:
Just a heads up for the upcoming changes in klipper regarding bed mesh, here is the message as displayed on mainsail:

>  Klipper changes to [bed_mesh] module
> In an upcoming change to Klipper, the default `bed_mesh` configuration will no longer be automatically loaded at startup. If you are using a `bed_mesh` configuration today then it is recommended to add an explicit `BED_MESH_PROFILE LOAD=default` command to your START_PRINT macro (or equivalent print start commands in your slicer). This change to Klipper is planned for February 1st 2023.

[link to the klipper announcement](https://github.com/Klipper3d/klipper/issues/5983)

Not sure if it also applies to adaptive bed mesh, but better safe than sorry!

jschuh, 2023-01-14:
Yep, saw that. I don't think it will be an issue, since the bed mesh macros are profile agnostic and just passthrough any additional arguments. If anything, I think this change will be entirely positive, since I've been bitten by accidentally saving a profile as a default (e.g. after a PID tune) and then having it automatically applied at startup.

That stated, I'll keep my eyes open to make sure the change is compatible.

closed 2023-01-14 by jschuh

flow23, 2023-01-15:
Hi, as far as I can see there is no explicit load of the bed mesh in your macros.

jschuh, 2023-01-19:
Yep, as I mentioned above the bed mesh macros are profile agnostic, because they just re-probe on each print (unless disabled by setting `variable_start_level_bed_at_temp: 0`). That stated, if one wanted to have a profile applied at startup they could just add something like `BED_MESH_PROFILE LOAD=default` to the `gcode` section of `[gcode_macro _km_options]`.

jschuh, 2023-01-23:
Figured I'd circle back and leave some notes here. Looking at [Arksine's PR](https://github.com/Klipper3d/klipper/pull/5877), he's changing only the profile loading behavior at Klipper initialization. Whereas [this is the part of the code](https://github.com/Klipper3d/klipper/blob/6026a99ad84065c7145b5aaf6f75ac332d93dc67/klippy/extras/bed_mesh.py#L603) that sets the default value of the `PROFILE` argument in `BED_MESH_CALIBRATE`.

So, it doesn't seem like there's anything that needs to change in these macros (at least not for the PR associated with the notification).