jschuh/klipper-macros #31: Macro for loading and unloading of filament is not working

flow23, 2023-01-13:
2:41 PM - Error evaluating 'gcode_macro _load_unload:gcode': TypeError: '<' not supported between instances of 'str' and 'int'
2:41 PM - load_filament LENGTH=150 SPEED=1500 EXTRUDER=extruder MINIMUM=250

I guess there is a problem with type conversion.

jschuh, 2023-01-14:
Interesting... I don't think the `SPEED` parameter had worked properly since I first checked it in. I guess that means you're the first person to actually use it. Anyway, it should be fixed now.

closed 2023-01-14 by jschuh

flow23, 2023-01-14:
Will try it next week. Thanks.

justlikeef, 2023-01-19:
Even though I got the error, it worked for me from the Mainsail web page, but never worked from the touchscreen interface on my Pi through KlipperScreen