jschuh/klipper-macros #34: LCD setup menu error related to bed mesh

scmanjarrez, 2023-01-19:
Do I need custom hardware to use your LCD menus? I'm using stock display from ender3 (12864)

```python                                                                                                  
Error evaluating 'menu __main __setup __bed_surface:input': ValueError: '' is not in list                                                                                                     
Traceback (most recent call last):                                                                                                                                                            
  File "/home/printer/klipper/klipper/klippy/extras/gcode_macro.py", line 61, in render                                                                                                       
    return str(self.template.render(context))                                                                                                                                                 
  File "/home/printer/klipper/venv/lib/python3.9/site-packages/jinja2/environment.py", line 1090, in render                                                                                   
    self.environment.handle_exception()                                                                                                                                                       
  File "/home/printer/klipper/venv/lib/python3.9/site-packages/jinja2/environment.py", line 832, in handle_exception                                                                          
    reraise(*rewrite_traceback_stack(source=source))                                                                                                                                          
  File "/home/printer/klipper/venv/lib/python3.9/site-packages/jinja2/_compat.py", line 28, in reraise                                                                                        
    raise value.with_traceback(tb)                                                                                                                                                            
  File "<template>", line 2, in top-level template code                                                                                                                                       
ValueError: '' is not in list
List drawing error
Traceback (most recent call last):
  File "/home/printer/klipper/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/printer/klipper/venv/lib/python3.9/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/printer/klipper/venv/lib/python3.9/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/printer/klipper/venv/lib/python3.9/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 2, in top-level template code
ValueError: '' is not in list

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/printer/klipper/klipper/klippy/extras/display/menu.py", line 625, in draw_container
    text = current.render_name(selected)
  File "/home/printer/klipper/klipper/klippy/extras/display/menu.py", line 164, in render_name
    name = str(self._render_name())
  File "/home/printer/klipper/klipper/klippy/extras/display/menu.py", line 66, in _render_name
    context = self.get_context()
  File "/home/printer/klipper/klipper/klippy/extras/display/menu.py", line 488, in get_context
    value = (self._eval_value(context) if self._input_value is None
  File "/home/printer/klipper/klipper/klippy/extras/display/menu.py", line 521, in _eval_value
    self._input_tpl.render(context)))
  File "/home/printer/klipper/klipper/klippy/extras/gcode_macro.py", line 66, in render
    raise self.gcode.error(msg)
gcode.CommandError: Error evaluating 'menu __main __setup __bed_surface:input': ValueError: '' is not in list
```
[printer.txt](https://github.com/jschuh/klipper-macros/files/10461245/printer.txt)
[macros.txt](https://github.com/jschuh/klipper-macros/files/10461246/macros.txt)

jschuh, 2023-01-19:
You don't have any bed surfaces defined in your `[gcode_macro _km_options]` section in `macros.cfg`. Adding a line like this will solve it by adding two available surfaces.

```
variable_bed_surfaces: ['smooth_1','texture_1']
```

I always set this up on my printers, so it looks like I don't have any error handling for this case. In retrospect, I should just add a default surface for when there's no custom config. I'll get to that sometime in the next day or so.

scmanjarrez, 2023-01-19:
Sorry, I'm new with klipper, what are bed_surfaces?

closed 2023-01-19 by scmanjarrez

jschuh, 2023-01-19:
Oh, it's not a Klipper thing. It's [described here](https://github.com/jschuh/klipper-macros#bed-surface),  and emulates a [corresponding Prusa feature](https://help.prusa3d.com/article/steel-sheet-profiles_1955). It just lets you maintain custom offsets for as many build surfaces as you want, and switch between them. It's most useful for printers with no probe or non-contact probes, but I find it helps even with contact probes (e.g. BLTouch) because you typically want a different level of squish between something like a smooth versus textured PEI sheet.