jschuh/klipper-macros #39: [BUG] Unable to start print due to "bed_surfaces" error

Will-wastelander, 2023-02-07:
**Describe the bug**
Please provide a clear and concise description of what the bug is.
When starting a print, I get the below error after bed_mesh is completed.
Printer is Ender 3 Pro w/ Duet 2 Wifi. Prints start w/o issue if klipper-macros is not loaded.

**Klipper error messages**
Paste any error Klipper error messages you observed when encountering the bug.
Error evaluating 'gcode_macro _apply_bed_surface_offset:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'bed_surfaces'

**Klipper config**
Paste any relevant sections from your Klipper config, or attach the full config if you prefer. Please also note any other macros you are including as part of your config.

printer.cfg has basics for functions.

[gcode_macro _km_options]
# These are examples of some likely customizations:
# Any sheets in the below list will be available with a configurable offset.
variable_bed_surfaces: ['smooth_1','texture_1']
# Length (in mm) of filament to load (bowden tubes will be longer).
variable_load_length: 250.0
# Hide the Octoprint LCD menu since I don't use it.
#variable_menu_show_octoprint: False
# Customize the filament menus (up to 10 entries).
#variable_menu_temperature: [
#  {'name' : 'PLA',  'extruder' : 200.0, 'bed' : 60.0},
#  {'name' : 'PETG', 'extruder' : 230.0, 'bed' : 85.0},
#  {'name' : 'ABS',  'extruder' : 245.0, 'bed' : 110.0, 'chamber' : 60}]
# Length of filament (in millimeters) to purge at print start.
variable_start_purge_length: 30 # This value works for most setups.
gcode: # This line is required by Klipper.
# Any code you put here will run at klipper startup, after the initialization
# for these macros. For example, you could uncomment the following line to
# automatically adjust your bed surface offsets to account for any changes made
# to your Z endstop or probe offset.
#  ADJUST_SURFACE_OFFSETS

# This line includes all the standard macros.
[include klipper-macros/*.cfg]
# Uncomment to include features that require specific hardware support.
# LCD menu support for features like bed surface selection and pause next layer.
#[include klipper-macros/optional/lcd_menus.cfg]
# Optimized bed leveling
[include klipper-macros/optional/bed_mesh.cfg]

# The sections below here are required for the macros to work.
[idle_timeout]
gcode:
  _KM_IDLE_TIMEOUT

jschuh, 2023-02-07:
Usually that means there's a mistake in the `[save_variables]` section of your config. Unfortunately the portion of your config that you provided doesn't include the `[save_variables]` section, so I can't be sure, but you can find the required config sections at the bottom of the [Klipper Setup portion of the readme](https://github.com/jschuh/klipper-macros#klipper-setup).

Also, please ensure you're running the most current Klipper version, as [I landed a somewhat recent Klipper change](https://github.com/Klipper3d/klipper/pull/5922) to help diagnose misconfigured `[save_variables]`.

Will-wastelander, 2023-02-07:
Looks like the variables.cfg file is empty. I copied the Klipper Setup section to my printer.cfg, I've included my complete printer.cfg this time.
```
[save_variables]
filename: ~/printer_data/variables.cfg # UPDATE THIS FOR YOUR PATH!!!
```

```
pi@E3Pro-mainsailos:~/printer_data $ ll variables.cfg
0 -rw-r--r-- 1 pi pi 0 Feb  4 20:44 variables.cfg
pi@E3Pro-mainsailos:~/printer_data $ cat variables.cfg
pi@E3Pro-mainsailos:~/printer_data $
```
printer.cfg
```
[include shell_command.cfg]
#[include mainsail.cfg]
#[include macros.cfg]
[include timelapse.cfg]
#[include klipper-macro.cfg]
#[include addons/inputshaper.cfg]
#[include addons/neopixel.cfg]
#[include addons/runout.cfg]
#[include addons/M20x.cfg]

# This file contains common pin mappings for Duet2 Eth/Wifi boards. To
# use this config, the firmware should be compiled for the SAM4E8E.

# See docs/Config_Reference.md for a description of parameters.

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100

[temperature_sensor mcu_temp]
sensor_type: temperature_mcu
min_temp: 0
max_temp: 100

[force_move]
enable_force_move: True

[mcu rpi]
serial: /tmp/klipper_host_mcu

# All customizations are documented in globals.cfg. Just copy a variable from
# there into the section below, and change the value to meet your needs.

[gcode_macro _km_options]
# These are examples of some likely customizations:
# Any sheets in the below list will be available with a configurable offset.
#variable_bed_surfaces: ['smooth_1','texture_1']
# Length (in mm) of filament to load (bowden tubes will be longer).
variable_load_length: 250.0
# Hide the Octoprint LCD menu since I don't use it.
#variable_menu_show_octoprint: False
# Customize the filament menus (up to 10 entries).
#variable_menu_temperature: [
#  {'name' : 'PLA',  'extruder' : 200.0, 'bed' : 60.0},
#  {'name' : 'PETG', 'extruder' : 230.0, 'bed' : 85.0},
#  {'name' : 'ABS',  'extruder' : 245.0, 'bed' : 110.0, 'chamber' : 60}]
# Length of filament (in millimeters) to purge at print start.
variable_start_purge_length: 30 # This value works for most setups.
gcode: # This line is required by Klipper.
# Any code you put here will run at klipper startup, after the initialization
# for these macros. For example, you could uncomment the following line to
# automatically adjust your bed surface offsets to account for any changes made
# to your Z endstop or probe offset.
#  ADJUST_SURFACE_OFFSETS

[... 312 lines cut ...]
    {action_respond_info(out|join("\n"))}

[delayed_gcode bed_mesh_init]
initial_duration: .01
gcode:
  BED_MESH_PROFILE LOAD=default

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 25.344
#*# pid_ki = 1.280
#*# pid_kd = 125.453
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 68.919
#*# pid_ki = 1.324
#*# pid_kd = 896.812
#*#
#*# [bltouch]
#*# z_offset = 1.384
```

Will-wastelander, 2023-02-07:
Here is the klipper.log of the failure, if it helps.

```
Mesh Bed Leveling Complete
save_config: set [bed_mesh default] version = 1
save_config: set [bed_mesh default] points =
  -0.087500, -0.036250, 0.063750, 0.101250, 0.192500, 0.203750, 0.241250
  -0.146250, -0.085000, 0.012500, 0.045000, 0.143750, 0.168750, 0.211250
  -0.128750, -0.071250, 0.017500, 0.062500, 0.152500, 0.178750, 0.213750
  -0.140000, -0.108750, -0.018750, 0.000000, 0.106250, 0.126250, 0.210000
  -0.166250, -0.072500, 0.022500, 0.075000, 0.166250, 0.202500, 0.237500
  -0.121250, -0.076250, 0.002500, 0.018750, 0.121250, 0.142500, 0.252500
  -0.071250, 0.003750, 0.106250, 0.137500, 0.230000, 0.258750, 0.290000
save_config: set [bed_mesh default] x_count = 7
save_config: set [bed_mesh default] y_count = 7
save_config: set [bed_mesh default] mesh_x_pps = 2
save_config: set [bed_mesh default] mesh_y_pps = 2
save_config: set [bed_mesh default] algo = bicubic
save_config: set [bed_mesh default] tension = 0.2
save_config: set [bed_mesh default] min_x = 30.0
save_config: set [bed_mesh default] max_x = 195.0
save_config: set [bed_mesh default] min_y = 30.0
save_config: set [bed_mesh default] max_y = 195.0
Bed Mesh state has been saved to profile [default]
for the current session.  The SAVE_CONFIG command will
update the printer config file and restart the printer.
Error evaluating 'gcode_macro _apply_bed_surface_offset:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'bed_surfaces'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/pi/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/klippy-env/lib/python3.9/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 3, in top-level template code
  File "/home/pi/klippy-env/lib/python3.9/site-packages/jinja2/environment.py", line 471, in getattr
    return getattr(obj, attribute)
jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'bed_surfaces'
Error evaluating 'gcode_macro _apply_bed_surface_offset:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'bed_surfaces'

Error evaluating 'gcode_macro _apply_bed_surface_offset:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'bed_surfaces'

Exiting SD card print (position 1361)
```

jschuh, 2023-02-07:
It sure seems like nothing is getting written to your `save_variables` file, since the macros should always write some state at startup. Could you try running this command in the Klipper console:

```
SAVE_VARIABLE VARIABLE=test VALUE='"test"'
```

Check immediately afterward to see if it's written to the file, and if there's anything in the Klipper log. The log message on a `save_variables` write failure should include the text: `Unable to save variable`

Will-wastelander, 2023-02-07:
Ran command, and get this.

```
pi@E3Pro-mainsailos:~ $ cat printer_data/variables.cfg
[Variables]
test = 'test'
```

jschuh, 2023-02-07:
Hm... that means `_INIT_SURFACES` just isn't running; or worse, `INIT_GLOBALS` either isn't running or isn't completing. That would actually break a lot of things, since it performs all the initialization for the macros.

Are you getting any Klipper console messages at startup? (I.e. after restarting in Mainsail via: Power button -> Klipper Control: Restart.)

If there are no errors in the console then maybe there's something about your system where the `initial_duration` on `INIT_GLOBALS` is too short. I haven't heard of anyone else hitting that issue, but I have observed that if the duration is too low Klipper won't fire the event. It's in `globals.cfg` and looks like this:

```
[delayed_gcode INIT_GLOBALS]
# This runs once at startup and initializes all macros.
initial_duration: 0.0001
```

You could try changing `initial_duration` to something larger like `0.01` and then restarting and seeing if anything new is written to the `variables.cfg`. If that solves it, I'll land a change loosening the initial duration.

Failing that, the most likely possibility is a conflict with some other macros you're loading. Unfortunately, that's the point where I would probably just give up. The Klipper macro system isn't really designed for cleanly integrating multiple sets of macros, and conflicts are common enough. I have been slowly renaming things with unique prefixes to reduce the chances of a breaking collision, but that's still no guarantee of a fix.

Also, I noticed that you have the `variable_bed_surfaces` commented out. That shouldn't be a problem, but you can try uncommenting, restarting Klipper, and seeing if it works.

Will-wastelander, 2023-02-07:
No change in the variables.cfg file being written to. I have tried with and w/o the variable_bed_surfaces_ commented out.

I just checked the logs, using a clean log file, and found an error about "max_extrude_only_distance" After looking at my printer.cfg I see that setting was missing. I added it, and now see the variable.cfg file being written to.

I will test and report back.

xyleth, 2023-02-07:
I had this exact problem.  In the end, the issue was max_extrude_only_distance being set shorter than the load_length specified in the macros, and I only found that out by scrolling through the klipper log.  No console errors were thrown.

Setting max_extrude_only_distance > load_length fixed the problem and the bed_surfaces dict was written to the variables file properly.

Will-wastelander, 2023-02-07:
I can confirm that adding max_extrude_only_distance to my [extruder] config has resolved the issue.

jschuh, 2023-02-07:
Wow, I just put a bad entry as a test for one of my configs and I'm not seeing startup errors anymore either. Something must have changed in Klipper or Mainsail. I'll have to dig into it and see.

That stated, you can directly run `_CHECK_GLOBALS` to throw an error on configuration issues. I'll add something to the readme on using it for troubleshooting, and maybe add it to `PRINT_START` just to be safe.

Sorry you had the hassle, but at least I should be able to save anyone else from having the same problem in the future.

Will-wastelander, 2023-02-07:
Helping a project that I use, is never a hassle. Even if it's a config
issue on my side, LOL. I'm just glad it was able to be resolved quickly.

Thanks for the quick responses!

closed 2023-02-07 by jschuh