jschuh/klipper-macros #41: [FEATURE REQUEST] Z_TILT_ADJUST before BED_MESH_CALIBRATE_FAST.

Will-wastelander, 2023-02-10:
I have issues w/ z_tilt on one of my printers. I have added Z_TILT_ADJUST just before the BED_MESH_CALIBRATE_FAST line. But update manager is saying it's dirty due to this change. Is there a way to have a variable that would allow this to be enabled ?

jschuh, 2023-02-11:
Yeah, I wrote a patch a while ago to do this for `Z_TILT_ADJUST` and `QUAD_GANTRY_LEVEL`, but I don't have any hardware that uses these configs. So, I didn't land it since I don't want to land anything without real world testing.

I just rebased the patch as posted it as [quad_and_tilt](https://github.com/jschuh/klipper-macros/tree/quad_and_tilt). If you can verify that it works I'd be willing to land it.

The other alternative is to just wrap `PRINT_START` or `BED_MESH_CALIBRATE` with something like this in your config:

```
[gcode_macro PRINT_START]
rename_existing: KM_PRINT_START
gcode:

  # Put macro code here to run before PRINT_START.

  KM_PRINT_START {rawparams}

  # Put macro code here to run after PRINT_START but before the print gcode
```

falcon939, 2023-02-13:
Hi!

Thank you for your effort. This is just what I still wanted to created.
I have tested the [quad_and_tilt](https://github.com/jschuh/klipper-macros/tree/quad_and_tilt) branch with a `Z_TILT_ADJUST` printer. (Ender 5+)

It works as I would expect. 
It will do the `Z-Tilt` before the `Bed mesh`.

jschuh, 2023-02-13:
Okay, I landed it and kept the quad gantry part in there since in this context it does basically the same thing as z tilt.

closed 2023-02-13 by jschuh