jschuh/klipper-macros #42: M300 macro fails if no `scale` parameter is defined in the `[output_pin beeper]` printer section

HitLuca, 2023-02-15:
I have a printer with a beeper available, and I can trigger it from my mainsail interface just fine. When calling the M300 command it fails, with the error

`Error evaluating 'gcode_macro m300:gcode': UndefinedError: 'dict object' has no attribute 'scale'`

The macro seems to be looking for a `scale` parameter in the beeper printer section, and if it's not defined the macro crashes. I would either update the README by mentioning that a `scale` paramter must be set or provide a default value for the parameter in the macro

jschuh, 2023-02-15:
That shouldn't be possible. The `scale` parameter is retrieved from [printer.configfile.settings](https://www.klipper3d.org/Status_Reference.html#configfile), so Klipper [default initializes `scale` to 1.0](https://github.com/Klipper3d/klipper/blob/0b918b357cb0c282d53cbdf59e1931a2054cd60a/klippy/extras/output_pin.py#L22). 

I just checked one of my printers with an `[output_pin beeper]` section and verified it doesn't set a `scale` value. Klipper just provides the 1.0 default and everything works correctly.

Are you using a Klipper fork, or is it possible some other macro is creating a conflict?

HitLuca, 2023-02-16:
I think the issue is related to the fact that I'm using an active beeper (non PWM) whereas the `scale` value can be set only if `pwm` is set to `True`

HitLuca, 2023-02-16:
I fixed the offending statement as follows:

original
```cfg
SET_PIN PIN=beeper VALUE={
        printer.configfile.settings["output_pin beeper"].scale * 0.5
} CYCLE_TIME={ 1.0 / S }
```

new
```cfg
SET_PIN PIN=beeper VALUE={
        printer.configfile.settings["output_pin beeper"].scale * 0.5 if 'scale' in printer.configfile.settings["output_pin beeper"] else 1.0
} CYCLE_TIME={ 1.0 / S }
```

jschuh, 2023-02-16:
Ah, I see. Klipper just [uses 1.0 internally](https://github.com/Klipper3d/klipper/blob/0b918b357cb0c282d53cbdf59e1931a2054cd60a/klippy/extras/output_pin.py#L26) as the scale if `pwm` is false, and doesn't add it to the `settings` dictionary. That left me wondering how it handles the `CYCLE_TIME` parameter, but it turns out [Klipper just ignores it](https://github.com/Klipper3d/klipper/blob/0b918b357cb0c282d53cbdf59e1931a2054cd60a/klippy/extras/output_pin.py#L67) in that case.

Anyway, it's any easy fix to just add a `scale|default(1.0)` to handle the non-PWM case, and I think I'll filter out the `CYCLE_TIME` argument as well, since it's not doing anything when it's not PWM.

jschuh, 2023-02-16:
Okay, should be fixed now. Honestly hadn't even occurred to me that some hardware would be using buzzers, since I just haven't encountered one on any of my printers.

closed 2023-02-16 by jschuh