jschuh/klipper-macros #44: ERROR evaluaton `gcode_macro check .....[BUG]

numpty9989, 2023-02-17:
**Describe the bug**
I installed the macro in the correct directory and called it in the printer. cfg and when I restarted it pops up with an Error I have absolutely no idea what that means... my klipper is literally stock except for configuring and tuning my printer so nothing else is conflicting with it...

**Klipper error messages**
Error evaluating `gcode_macro check_km_config:gcode`: gcode.CommandError:park_y is invalid

**Klipper config**
My config is stock board config with tuned features like pressure advance PID tuned etc.... no other macros installed except PID hotend and bed...

jschuh, 2023-02-18:
This means the Y parking position is outside the bounds declared as `position_min` and `position_max` in the `[stepper_y]` section of the Klipper config. So, either the Klipper config is incomplet/invalid or `variable_park_y` in `[gcode_macro _km_options]` is set to an invalid Y coordinate.

Please ensure you have a working (i.e. printing) Klipper installation prior to attempting to configure these macros and have followed the [troubleshooting guidance](https://github.com/jschuh/klipper-macros#troubleshooting) in the README.

closed 2023-02-18 by jschuh

numpty9989, 2023-02-18:
Yes my config is fully working. I have had it working for weeks. I wanted to see all features that this adds. I will look into that and see where I went wrong maybe I had added extra digit or something. Cheers for updating.