jschuh/klipper-macros #46: [BUG] Error evaluating 'gcode_macro bed_mesh_calibrate_fast

MJeka, 2023-02-18:
After uploading the file to print I get the following error.

Error evaluating 'gcode_macro bed_mesh_calibrate_fast:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro bed_mesh_calibrate'

I add in Slicer 

BED_MESH_PROFILE LOAD=default

What does this error mean and how to fix it?

jschuh, 2023-02-18:
No config information was included in this report and this message could result from several potential configuration errors.

Please ensure you have a working (i.e. printing) Klipper installation prior to attempting to configure these macros and have followed the [troubleshooting guidance](https://github.com/jschuh/klipper-macros#troubleshooting) in the README.

closed 2023-02-18 by jschuh

MJeka, 2023-02-19:
After I commented out the lines in the file the error disappeared.

    # {% if km.start_level_bed_at_temp %}
    #   BED_MESH_CALIBRATE_FAST{% if MESH_MIN %} MESH_MIN={MESH_MIN}{% endif
    #                        %}{% if MESH_MAX %} MESH_MAX={MESH_MAX}{% endif %}
    # {% endif %}

jschuh, 2023-02-20:
I wouldn't advise arbitrarily commenting out parts of the macros over simply finding and fixing whatever mistake you made in your config, but that's up to you.

mpbecker, 2023-02-20:
I have the same problem here. But to me it seems to be a misunderstanding of how to deactivate the fast probing method. From the documentation I understand to not include the `bed_mesh.cfg` would be enough but obviously that is not the case because as soon as I do not include it, I get the same error as @MJeka

jschuh, 2023-02-20:
I'd like to help, which is why the bug reporting template explicitly asks for the Klipper config, and I then reiterated the need for said config in my first reply

In this particular case the macro initialization should [detect that no `bed_mesh` is configured, and set `start_level_bed_at_temp` to `False`](https://github.com/jschuh/klipper-macros/blob/35fd436e7d12a88f3694448187a20b19e9527d15/globals.cfg#L127). So, the fact that this error is even being encountered could point to any number of configuration errors: an invalid `bed_mesh` may be specified; `start_level_bed_at_temp` may be erroneously overridden to the wrong value; or the macro initialization may be failing for any number of reasons.

The thing is that I can't help diagnose and fix anything without seeing the content of the Klipper config.

mpbecker, 2023-02-20:
Sure, makes total sense. Here is my config that throws me the mentioned error. Klipper firmware is 0.11.

[printer.txt](https://github.com/jschuh/klipper-macros/files/10787423/printer.txt)
[macros.txt](https://github.com/jschuh/klipper-macros/files/10787424/macros.txt)

jschuh, 2023-02-20:
Okay, you have a valid `bed_mesh` section, so you shouldn't be getting any errors in `PRINT_START` (which is different than what @MJeka reported).

Is your issue just that you don't want `PRINT_START` to automatically perform a `BED_MESH_CALIBRATE_FAST` on each run? If that's the case you can simply add `variable_start_level_bed_at_temp: False` into your `[gcode_macro _km_options]` to disable that feature.

All of the override settings that affect `PRINT_START` are [explained in detail here](https://github.com/jschuh/klipper-macros#print-start-and-end).

mpbecker, 2023-02-20:
Sorry, I missed the `variable_start_level_bed_at_temp` in the documentation! That solved it for me, thank you!

MJeka, 2023-02-21:
> I wouldn't advise arbitrarily commenting out parts of the macros over simply finding and fixing whatever mistake you made in your config, but that's up to you.

I'll be happy to give more information and help in any way I can, I just don't understand what I need to do. I had to comment out this code so that I could print.

mpbecker, 2023-02-21:
If you post your configuration files like me, @jschuh can have a look if you missed something in your config.

MJeka, 2023-02-21:
> If you post your configuration files like me, @jschuh can have a look if you missed something in your config.

Change variable_start_level_bed_at_temp from True on False?

MJeka, 2023-02-21:
@jschuh My config files.
[klipper-macros.txt](https://github.com/jschuh/klipper-macros/files/10795689/klipper-macros.txt)
[printer.txt](https://github.com/jschuh/klipper-macros/files/10795690/printer.txt)
[globals.txt](https://github.com/jschuh/klipper-macros/files/10795691/globals.txt)

MJeka, 2023-02-21:
Change variable_start_level_bed_at_temp to false, error remains.

jschuh, 2023-02-21:
@MJeka, the files you provided show even more alterations of the macro source files. I'm sorry, but I can't help you, because I have no way of knowing what you're actually running. And the error you're reporting simply should not be possible if you had followed [the installation instructions](https://github.com/jschuh/klipper-macros#installation).

This is all open source code and I'm glad that people can customize their own forks to meet their needs (as many people have), but it's just not practical for me to try to troubleshoot issues in forks.

MJeka, 2023-02-22:
@jschuh I did everything according to the instructions. Ok, thanks for the macros.