jschuh/klipper-macros #47: [BUG] ADJUST_SURFACE_OFFSETS IGNORE malformed command

Will-wastelander, 2023-02-20:
**Describe the bug**
Please provide a clear and concise description of what the bug is.
When attempting to run `ADJUST_SURFACE_OFFSETS IGNORE` I get a malformed  command error.

**Klipper error messages**
Paste any error Klipper error messages you observed when encountering the bug.
`10:05 AM
Malformed command 'ADJUST_SURFACE_OFFSETS ignore'
10:05 AM
ADJUST_SURFACE_OFFSETS ignore
10:03 AM
Malformed command 'ADJUST_SURFACE_OFFSETS IGNORE'
10:03 AM
ADJUST_SURFACE_OFFSETS IGNORE
10:03 AM
Unknown command:"ADJUST_SURFACE_OFFSETS0"
10:03 AM
ADJUST_SURFACE_OFFSETS 0
10:03 AM
Malformed command 'ADJUST_SURFACE_OFFSETS IGNORE'
10:03 AM
ADJUST_SURFACE_OFFSETS IGNORE
10:02 AM
Active bed surface: default; offset: 0.744
10:02 AM
Z probe offset or endstop position changed. Run ADJUST_SURFACE_OFFSETS to adjust the offset for all saved surfaces by the change differential, or run ADJUST_SURFACE_OFFSETS IGNORE to hide this message without making changes.
10:01 AM
Klipper state: Disconnect
10:01 AM
SAVE_CONFIG
10:01 AM
bltouch: z_offset: 0.475
The SAVE_CONFIG command will update the printer config file
with the above and restart the printer.
10:01 AM
Z_OFFSET_APPLY_PROBE
9:25 AM
Active bed surface: default; offset: 0.744
9:25 AM
- Type HELP to get a list of available commands.
- Click on the "?" button to get a searchable list.
- Commands in the console are clickable and will be placed into the input field.
- Use the tab key to complete your inputs. If there are several options, a list is displayed.
- Use the ⇵ arrow keys to navigate through the previous entries.
`

**Klipper config**
Paste any relevant sections from your Klipper config, or attach the full config if you prefer. Please also note any other macros you are including as part of your config.
-- variables.cfg --
`[Variables]
bed_surfaces = {'active': 'default', 'available': {'default': {'offset': 0.744}}, 'endstop_z': 0.0, 'probe_z': 1.219}
`
-- printer.cfg --
`[include timelapse.cfg]
#[include addons/neopixel.cfg]
#[include addons/runout.cfg]

# This file contains common pin mappings for the BIGTREETECH Manta E3EZ
# To use this config, the firmware should be compiled for the
# STM32G0B1 with a "8KiB bootloader" "8 MHz crystal"
# and "USB (on PA11/PA12)" or "CAN bus (on PB12/PB13)".

# See docs/Config_Reference.md for a description of parameters.

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100

[temperature_sensor mcu_temp]
[... 348 lines cut ...]
    {action_respond_info(out|join("\n"))}

[delayed_gcode bed_mesh_init]
initial_duration: .01
gcode:
  BED_MESH_PROFILE LOAD=default

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 31.050
#*# pid_ki = 1.784
#*# pid_kd = 135.067
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 68.251
#*# pid_ki = 1.143
#*# pid_kd = 1018.640
#*#
#*# [bltouch]
#*# z_offset = 0.475
`

jschuh, 2023-02-20:
This is just how the Klipper command processor works. Only legacy gcode (e.g. M##, G##) can accept bare arguments. Everything else must be of the form `ARG=XX`. So, the correct form of the command is `ADJUST_SURFACE_OFFSETS IGNORE=1`

Will-wastelander, 2023-02-24:
This would this be a cosmetic bug then? Since the output is saying to run 'ADJUST_SURFACE_OFFSETS IGNORE' instead of 'ADJUST_SURFACE_OFFSETS IGNORE=1'

jschuh, 2023-02-24:
Ah, just a typo in the output text. Should be fixed now.

closed 2023-02-24 by jschuh