jschuh/klipper-macros #48: [BUG] Print fails using raft supports

rado79, 2023-02-20:
**Describe the bug**
I slice my models with Cura 5.1
Using supports with brim or skirt works a s aspected.
If i choose Raft (in my Case 4 Layers) i get this error and print fails.

**Klipper error messages**
00:30
SDCARD_RESET_FILE cannot be run from the sdcard
00:30
Error on 'SET_PRINT_STATS_INFO CURRENT_LAYER="-4"': CURRENT_LAYER must have minimum of 0
00:30
Error on 'SET_PRINT_STATS_INFO CURRENT_LAYER="-4"': CURRENT_LAYER must have minimum of 0

**Klipper config**
[config (2).zip](https://github.com/jschuh/klipper-macros/files/10787927/config.2.zip)
[CFFFP_mandibula.zip](https://github.com/jschuh/klipper-macros/files/10787982/CFFFP_mandibula.zip)

jschuh, 2023-02-21:
Thanks, this is useful. People have filed bugs about Cura passing negative layer heights, but I never knew why (I don't use Cura) and no one seemed to know what settings were the cause. Now that I know it's because Cura uses negative layer values for rafts, I can think of a better way to handle it.

jschuh, 2023-02-21:
Please let me know if bfe55c5 fixes the issue.

rado79, 2023-02-22:
No, same issue again.

Do we need this semicolon?
[image]

jschuh, 2023-02-22:
Even if it's not fixed you shouldn't be getting exactly the same error. Could you paste in the exact error text? Also, sorry if this is a dumb question, but I just need to verify you did restart Klipper after updating klipper-macros and attempting the print again?

And yes, that semicolon is required. The gcode you attached shows that the substitutions are all working correctly. The problem is just that Cura uses negative layer values for the raft.

rado79, 2023-02-22:
[image]

rado79, 2023-02-22:
This is while purge line prints
[image]
It shows Layer 2 of 774

rado79, 2023-02-22:
After purge line, the print stops with the error
[image]
shows Layer 112 of 774

rado79, 2023-02-22:
And i restartet everything again one more time, same error

rado79, 2023-02-22:
I dont know if my update process i the right thin i do. I copy the text from layers.cfg of your commit,
and paste it in my Klipper config to layers.cfg

rado79, 2023-02-22:
[layers.zip](https://github.com/jschuh/klipper-macros/files/10805783/layers.zip)

rado79, 2023-02-22:
> Oh, could you just do a `git pull` to update so I can be sure it's running the right version? You may have to do a `git reset --hard HEAD` first to revert any local changes to the macros.

if i do so, do i lose changes like "G1 E12 F1000" direct before purgeline begins?

jschuh, 2023-02-22:
Yes, it would revert your local changes. The thing is that I can't really figure out what's going on if the macros aren't a stock installation of the most current version.

rado79, 2023-02-22:
okay, i wil do it like you said before

rado79, 2023-02-22:
[image]

jschuh, 2023-02-22:
I just landed 54b137d, which simply ignores negative layer values. The previous change tried to do some extra validation, but I decided that just isn't worth the trouble.

jschuh, 2023-02-22:
I'd been thinking about adding a config option for user-supplied gcode immediately before the purge/print starts, since I noticed that's a common customization. I'll try to test and land that today.

rado79, 2023-02-22:
After a hard reset and update to the latest commit, Raft prints as expected.
Layer counter starts with first layer of the model.
Well done!

jschuh, 2023-02-22:
Ah, interesting. I'd thought about adding purge priming like Prusa printers do, but I hadn't needed it on any of my printers. It's an easy change to incorporate into `DRAW_PURGE_LINE`, so I'll look into it.

That stated, I still think generic user-defined gcode before the print is independently needed for things like docking probes and the ERCF.