jschuh/klipper-macros #50: [BUG] Internal error during connect: unexpected EOF while parsing

Reconos87, 2023-02-24:
Before making a report please ensure you've followed the troubleshooting steps at:
https://github.com/jschuh/klipper-macros#troubleshooting

**Describe the bug**
Upon starting Mainsail i get the following error:

"Klipper reports: ERROR

Internal error during connect: unexpected EOF while parsing (<unknown>, line 0)

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted"

If i remove [include klipper-macros/*.cfg] the Printer doesn't display any errors, except for the basic macros missing. 
Was working fine until i updated this morning. I have removed the entire klipper-macros and re-pulled it.

**Klipper errors**
No commands fired except for "startup" or what the initialization does when the printer is turned on.

**Klipper config**
Attached Klippy log with configs.
[klippy (3).log](https://github.com/jschuh/klipper-macros/files/10824161/klippy.3.log)

jschuh, 2023-02-24:
Very strange. I was able to repro this on an Armbian SBC, but nowhere else. It's failing inside the Python ast library when parsing a variable declaration in `globals.cfg`. That certainly implies that the issue is a quirk of a specific port of Python and/or the ast library. Simply changing the declaration from a double-quoted empty string to a single quoted empty string seems to solve the problem.

I'll post a patch to fix it momentarily.

jschuh, 2023-02-24:
Okay, it should be fine now. Very odd bug. I guess I'll have to remember that Python's ast library can sometimes choke on empty double-quoted string literals.

closed 2023-02-24 by jschuh