jschuh/klipper-macros #53: Add support for heater-less chambers

HitLuca, 2023-03-03:
I finally had the time to install a chamber thermistor on my printer, without a dedicated heater. My printing process now is as follows:
- preheat bed to 100C
- preheat nozzle to 169C
- start my nevermore air filter to circulate air around
- manually check the printer after a while and wait until my chamber temp reaches X degrees

Now, the preheating and nevermore spinup is all done through a temperature preset, so it's not really an issue for now, but what could be improved in the print start macros is add support for heater-less chambers, where the user only uses the bed for heating them.

To me this would require: a preheat temp for nozzle and bed, which will be different form actual printing temps (for example I chose 169 for the extruder since it doesn't start melting filament and the hotend fan still moves heat around), spin up chosen auxiliary fans (one nevermore filter in my case but I also heard people spinning up par cooling fans or two nevermores for more air circulation). Wait for the chamber thermistor to reach the desired chamber temperature, using  a simple `TEMPERATURE_WAIT` command and then start printing. Chamber commands won't work since you don't have a proper chamber element in your printer.cfg, so it needs to be its own.

Let me know if it sounds feasible, happy to help if needed

HitLuca, 2023-03-04:
Just so I don't create a new issue, it would be nice that the print start macro turns off part cooling fans, as slicers don't usually do it themselves and the user may already have them on. Noticed this issue when trying to speed up chamber preheating by turning on pc fans

Novacat170, 2023-03-06:
I would also love for the print start macro (or part of it) gcode to be called from another cfg file. every time there is an update the "changes" i did to it like calling for led effects during probing are lost... maybe I'm not doing it right but coding was never my good point (I'm more of a mechanical thingies guy),

HitLuca, 2023-03-06:
@Novacat170 you are replacing the changes you made with the updated version on this repo, you can just merge them instead to have both your changes and the recent upstream changes. Check out the basics of git commands to see how to do that

jschuh, 2023-03-09:
I'll think about it, but my concern is that hardware configs would be so different that it'd be hard to come up with anything  sufficiently generic.

I should mention that I did experiment with a hacky way to do something like this. I configured a fan under the bed heater as `[heater_generic chamber]` in `printer.cfg`, such that the `heater_pin` was actually the control pin for the fan, and the thermistor was just setup normally. That way the fan pushed more air across the heater bed when the chamber was "heating", which warmed the chamber as a whole.

There are caveats to that approach. PID tuning required a bit of manual adjustment, and it needed extra leeway in the corresponding `verify_heater` sections for both the bed header and the fan "heater". The biggest issue though was that it only worked when the bed was within the appropriate range for the chamber's target temperature (which depended on a variety of factors, including the size of the enclosure, how well it was insulated, and the ambient temperature in the room).

If I recall correctly, when printing ABS on a 100C bed I could never get the chamber much higher than 55C, or lower than 40C. So, I had to stick within that range to avoid a heater error, but it did get the enclosure significantly hotter and the temperature inside was much more stable than without it.

I did think about how to make that approach more reliable, but didn't keep the setup long enough to get around to it.

closed 2023-05-11 by jschuh