jschuh/klipper-macros #56: Unusual (intermittent) error while using Z-TILT and bed_mesh_fast in Print Start

TannerPat, 2023-03-15:
Let me start by saying congratulations on an amazing collection of Macros,
I love the At Layer/ Next Layer macros, it allows for so many options, keep up the great work.

I have my Printer.cfg set to use the temps provided from slicer,
Heat Bed and preheat Extruder, Run Z-Tilt to sync both stepper motors, do a fast bed mesh calibrate, 
finish heating extruder and start printing.
It worked perfectly for 10+ prints yesterday, pausing at layer X to do a filament swap, changing settings at layers,
I was delighted with life and thought I found my ideal solution, but the gremlins got in last night while I slept,

I am using a BLTouch for vitrual Z-Endstop and have noticed some intermittent Errors cropping up.

First print this morning I got a BLTouch failed to deploy Error.
I restarted everything (complete power-down and reboot) 
BLTouch ran its self test ok and next print ran without issue. 

Several (successful) prints later I got another BLTouch failed to deploy Error.
But manually checking with BLTOUCH_Debug reveals no issues.
Console output:
13:43 File opened:First layer calibration_0.2mm_PETG_I3MEGAS_5m.gcode Size:132373
13:43 File selected
13:47 BLTouch failed to deploy
13:47 BLTouch failed to deploy
13:47 BLTouch failed to deploy
13:47 BLTOUCH_DEBUG COMMAND=pin_down
13:47 Sending BLTOUCH_DEBUG COMMAND=pin_down
13:47 BLTOUCH_DEBUG COMMAND=pin_up
13:47 Sending BLTOUCH_DEBUG COMMAND=pin_up

Once I reset and try again it homes, then probes Z after Bed Heating, 
then I have received errors during Z-Tilt
See Console Output below:
13:49 Retries: 10/20 Probed points range: 0.006250 tolerance: 0.002500
13:49 probe at 0.000,125.000 is z=2.545837
13:49 probe at 145.000,125.000 is z=2.554587
stepper_z = -0.010584
stepper_z1 = -0.001845
13:49 Making the following Z adjustments:
13:49 Retries: 11/20 Probed points range: 0.008750 tolerance: 0.002500
13:49 Retries aborting: Probed points range is increasing.

It appears the wrong results are getting logged, If I immediately manually re-run a Z-Tilt
I get the following:
13:49 Z_TILT_ADJUST
13:49 probe at 0.000,125.000 is z=2.551834
13:49 probe at 145.000,125.000 is z=2.554334
13:49 Making the following Z adjustments:
stepper_z = -0.003690
stepper_z1 = -0.001200
13:49 Retries: 0/20 Probed points range: 0.002500 tolerance: 0.002500

If I re-run after doing a manual Z-Tilt, it will happily probe the bed for the bed_mesh_fast and complete the print:
13:51 Generating new points...
13:51 bed_mesh: generated points
Index | Tool Adjusted | Probe
13:51 0 | (1.0, 24.0) | (30.0, 5.0)
13:51 1 | (43.5, 24.0) | (72.5, 5.0)
blah ......
blah ......
13:51 23 | (128.5, 209.0) | (157.5, 190.0)
13:51 24 | (171.0, 209.0) | (200.0, 190.0)
13:52 Mesh Bed Leveling Complete
13:52 Bed Mesh state has been saved to profile [default]
for the current session. The SAVE_CONFIG command will
update the printer config file and restart the printer.
13:52 Active bed surface: default; offset: 0.000
13:57 Done printing file

Config File:
[printer.cfg.txt](https://github.com/jschuh/klipper-macros/files/10980949/printer.cfg.txt)

The only possible difference I can think of is the bed was already at temp when I got the last error, (can't remember what setting this morning during first error), is there some sort of timing issue with macro calling the BLTouch to action?

Thanks for any help / suggestions

TannerPat, 2023-03-15:
Also uploaded the Klippy log,
The first BLTouch failure is at **Line 2423**, followed by the first Printer restart at **Line 2521**

Second BLTouch failure at **Line 14264**, 
My Debugging of BLTouch at Lines 14346 - 14359, and restarting print at Line 14405.
Z-Tilt begins at Line 14597, and aborts at **Line 14671**.
Manual Z-Tilt calibrated at **Line 14739** shows within tolerance.
Restarted Klipper at Line 14982

Ran a successful print starting Line 17725, Z-Tilt at Line 17972 - 18042, Bed Mesh Probe at Line 18047-18109, 
Print Finished at Line 29378.

Next Print starts Line 32196,
Third BLTouch failure at **Line 32143**, manual BLTouch Debug at Line 32184, and macro Z-Tilt aborted at Line 32419.
Manual Z-Tilt (Line 32442) showed that probed range was at tolerance (0.0025) 
and not the 0.00875 reported during macro Z-Tilt.

 
[klippy.log](https://github.com/jschuh/klipper-macros/files/10981468/klippy.log)

jschuh, 2023-03-15:
Thanks for the very thorough report. From the logs it seems pretty clear that the issue you're encountering is unrelated to the macros, and is almost certainly a hardware failure.

My guess would be that you have a bent/misaligned probe pin (that's what it's always been when I've seen this error). The pin doesn't need to be off by much to cause it to sporadically fail to deploy, and even a small movement of the printhead can often loosen it up again and make it appear to be fine for a bit.

What I've done when I encounter this issue is just take the pin out and inspect it to see if it's bent; you can visually inspect, see how it rolls on a flat surface, etc. Usually I can just bend a misaligned pin back into shape and the probe starts working fine, but if it's bad enough you may have to replace the pin (IIRC I've had to do that once on a BLTouch and once on a clone). If the pin doesn't look bent it may just be that it had shifted enough in the probe body that it was no longer seating correctly, and simply reassembling the probe properly will get everything back into alignment.

There are a number of guides/videos out there on how to to troubleshot and/or replace the probe pin in a BLTouch or clone. You can just follow those to learn how to disassemble and inspect the probe.

I should add that there are also certainly other potential causes for the failure, such as a break/short in the wiring or the hall sensor in the probe falsing due to the heat of the bed. But for me it's always been the probe pin.

closed 2023-03-15 by jschuh

TannerPat, 2023-03-15:
Thanks for the quick response Justin,
I am not certain, but I don't think it's an issue with the pin, I installed a new pin following an unplanned bed strike 😅  a few days ago.
The bed heat does sound like a possible culprit, I vaguely recall something about BLTouch not being very reliable above 60 degrees, Is there a way to edit your macro to limit bed temp to 55° or some such to facilitate probing with BLTouch.

jschuh, 2023-03-15:
Looking at the log, it's almost certainly not the probe's hall sensor falsing due to thermal drift. That sort of thing would typically cause a small drift in the Z offset, rather than an outright failure. But more importantly, the log shows that some of the failures occurred during the initial homing, when the bed and extruder were both at room temperature:

```
Stats 125868.8: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000002 mcu_task_stddev=0.000002 bytes_write=9228 bytes_read=38759 bytes_retransmit=9 bytes_invalid=0 send_seq=740 receive_seq=740 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=18 freq=400013693 sd_pos=33202 heater_bed: target=80 temp=23.3 pwm=1.000 sysload=0.75 cputime=12.912 memavail=682348 print_time=44720.127 buffer_time=0.000 print_stall=0 extruder: target=235 temp=26.4 pwm=1.000
BLTouch failed to deploy
```

Is it possible you didn't seat something quite right when you replaced the pin?

The two big things are that the pin moves smoothly and the top set screw is at the right depth. The screw depth is actually very important, since it's the ferous core of the magnet that drives the pin, so it needs to be centered properly in the coil; if it's too low or too high the probe won't deploy reliably.

Failing that, my next best guess is one of the connectors was left a bit loose after replacing the pin (or a break/short, but that seems less likely if it was working fine before replacing the pin).

TannerPat, 2023-03-16:
Thanks again for the update, midnight here so will have a close look at the replaced pin in the morning. 
Do you think it's OK for the BLTouch to be probing a bed that is stable at 80° (PETG) prints. Prior to your macros I used to manually preheat to 50/ 55°C and probe. Not perfect with extra thermal movement, but it never created any issues. Bed mesh was typically in the region of 0.1 mm 
Regards 
Pat

jschuh, 2023-03-16:
Shouldn't be a problem. I regularly use BLTouch/clone probes for printing ABS and hotter filaments in enclosures without any issues. That's with the bed at 100C-110C and the enclosure at 40C-55C. I've done a few tests to determine the drift, and it seems to be on the order of 0.01mm, so I haven't had any problems.

You definitely want the bed stabilized at the target temperature before probing, because the exact shape of the bed can shift enough as it heats up that you can get unreliable first layers. You can demonstrate this just by comparing the images of mesh maps at e.g. at 50C vs 80C.

TannerPat, 2023-03-16:
Yeah, I have noticed a significant improvement in bed mesh with the macro's, typically getting 0.1mm range now  ompared to .2+ before.
Thanks for all the feedback and again great work with the macro collection 👍

TannerPat, 2023-03-17:
Found the solution: See below for anybody having similar issues:
After dismantling the BlTouch **several** times, carefully checking and cleaning the moving pin and reassembling I was still having the same intermittent issue. Did the usual google search etc.
The issue is the little set(grub) screw which retains the pin.
This **should not be fully tightened** - No mention of this anywhere in the manual.
If fully tightened down the magnet inside the pin holds on too tightly causing pin-down issues.
You need to carefully back off this set-screw until the pin-down gremlin disappears.
**Do this in small steps** - if you reverse the set screw too far the pin can drop unexpectedly during a print

jschuh, 2023-03-17:
Sorry you had so much trouble with it. I thought that might be the issue, and tried to call it out earlier (although I guess this explanation could have been clearer):

> The screw depth is actually very important, since it's the ferous core of the magnet that drives the pin, so it needs to be centered properly in the coil; if it's too low or too high the probe won't deploy reliably.

TannerPat, 2023-03-17:
> [EXTERNAL] This email was sent from outside of UCC.

No worries, it was a bit of a process to find the cause but sorted now 😊
and thanks for your help

Regards
Pat