jschuh/klipper-macros #58: Load existing bed mesh instead of probing

raechten, 2023-03-18:
Apart from overriding PRINT_START, is there a way to load an existing bed mesh instead of probing the bed before each print? Especially with shorter prints, this greatly reduces the overall time

jschuh, 2023-03-18:
Overriding `PRINT_START` is quite easy; [the docs have a template you can copy](https://github.com/jschuh/klipper-macros#print-start-and-end). You could also duplicate the old Klipper behavior by loading the profile in the `gcode` for your `_km_options`, which runs at Klipper startup. Or you could just load it in `_km_options` `variable_start_gcode_before_print` before each print.

closed 2023-03-18 by jschuh

jschuh, 2023-05-13:
There's a [suggestion in the Ideas discussion](https://github.com/jschuh/klipper-macros/discussions/96) of tying this to the bed sheet selection, which I'm thinking about doing.