jschuh/klipper-macros #62: [BUG] Sonic Pad Klipper config with klipper macros

KaratsuFr, 2023-04-03:
**Describe the bug**
I'm trying to use those great macro on my newly Sonic Pad from Creatly. I have 2 errors 1 on macro bed_mesh_calibrate_fast and if i'm removing the bed_mesh then i have a macro on _km_layer_run:gcode

**Klipper errors**

```
!! {"code":"key165", "msg": "Error evaluating 'gcode_macro bed_mesh_calibrate_fast:gcode': UndefinedError: 'instance object' has no attribute 'gcode_macro bed_mesh_calibrate'
", "values": ["gcode_macro bed_mesh_calibrate_fast:gcode", "UndefinedError: 'instance object' has no attribute 'gcode_macro bed_mesh_calibrate'
"]}
```

-- if no mesh 
```
!! {"code":"key165", "msg": "Error evaluating 'gcode_macro _km_layer_run:gcode': UndefinedError: 'dict object' has no attribute 'info'
", "values": ["gcode_macro _km_layer_run:gcode", "UndefinedError: 'dict object' has no attribute 'info'
"]}
```

**Klipper config**
code after the display status section:
```

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
  TURN_OFF_HEATERS
  {% if "xyz" in printer.toolhead.homed_axes %}
    G91
    G1 Z4.5 F300
    G90
  {% else %}
    {action_respond_info("Printer not homed")}
  {% endif %}
    G28 X Y
  {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
    G1 Y{y_park} F2000
    M84
  CANCEL_PRINT_BASE
  
# Do nozzle priming
[gcode_macro PRIME_EXTRUDER]
gcode:
    M117 Priming
    G92 E0
    G1 X10.1 Y20 Z2.0 F2000
    G1 X10.1 Y200 Z0.28 F2000.0 E15; Draw the first line
    G1 X10.4 Y200 Z0.28 F5000.0; Move to the side a little
    G1 X10.4 Y20.0 Z0.28 F2000.0 E30; Draw the second line
    G92 E0 
    G1 F2400 E-0.7 # Small retract to remove pressure
    G1 Z2.0 F3000 ;Move Z Axis up

[stepper_x]
step_pin: PC2
dir_pin: PB9
enable_pin: !PC3
microsteps: 16
rotation_distance: 39.92
endstop_pin: ^PA5
position_endstop: 0
position_max: 245
homing_speed: 50

[stepper_y]
step_pin: PB8
dir_pin: PB7
enable_pin: !PC3
microsteps: 16
rotation_distance: 39.56
endstop_pin: ^PA6
position_endstop: 0
position_max: 225
homing_speed: 50

[stepper_z]
step_pin: PB6
dir_pin: !PB5
enable_pin: !PC3
rotation_distance: 8.35
microsteps: 16
endstop_pin: probe:z_virtual_endstop           #enable to use bltouch
# endstop_pin: ^PA7              #disable to use bltouch
# position_endstop: 0
position_min: -10
position_max: 255
homing_speed: 4
second_homing_speed: 1
homing_retract_dist: 2.0

[extruder]
max_extrude_only_distance: 1000.0
step_pin: PB4
dir_pin: PB3
enable_pin: !PC3
microsteps: 16
rotation_distance: 34.406
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[verify_heater extruder]
check_gain_time: 200
hysteresis: 5

[heater_bed]
heater_pin: PA2
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA0

[mcu]
serial: /dev/serial/by-id/usb_serial_1
restart_method: command

# [mcu rpi]
# serial: /tmp/klipper_host_mcu

# [adxl345]
# cs_pin: rpi:None
# spi_speed: 2000000
# spi_bus: spidev2.0

# [resonance_tester]
# accel_chip: adxl345
# accel_per_hz: 70
# probe_points:
#     117.5,117.5,10

[input_shaper]
#shaper_type_y = 2hump_ei
#shaper_freq_y = 52.6
#shaper_type_x = mzv
#shaper_freq_x = 62.8

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 5000
max_z_velocity: 10
max_z_accel: 1000
square_corner_velocity: 5.0

[bltouch]
sensor_pin: ^PB1       #signal check port ^stand for pull up
control_pin: PB0       #singal control prot
x_offset: -50.0
y_offset: -5.0
#z_offset: 0          #z off_set configuration
speed: 20
stow_on_each_sample = false #high speed for bltoch,
samples: 1
#probe_with_touch_mode = true

[safe_z_home]
home_xy_position:160,120
speed: 200
z_hop: 10
z_hop_speed: 10

[bed_mesh]
speed: 150
mesh_min: 15,30         #need to handle head distance with bl_touch
mesh_max: 190,190       #max probe range
probe_count: 5,5
fade_start: 1
fade_end: 10
fade_target: 0
algorithm: bicubic

[bed_screws]
screw1: 25, 33
screw2: 202, 33
screw3: 202, 202
screw4: 25, 202

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 0.100
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.022500, -0.040000, -0.015000, -0.102500, -0.205000
#*# 	-0.012500, -0.017500, -0.022500, -0.082500, -0.217500
#*# 	-0.020000, -0.045000, -0.012500, -0.087500, -0.210000
#*# 	-0.035000, -0.037500, -0.002500, -0.080000, -0.170000
#*# 	-0.002500, 0.007500, 0.032500, -0.047500, -0.165000
#*# tension = 0.2
#*# min_x = 15.0
#*# algo = bicubic
#*# y_count = 5
#*# mesh_y_pps = 2
#*# min_y = 30.0
#*# x_count = 5
#*# max_y = 190.0
#*# mesh_x_pps = 2
#*# max_x = 190.0
#*#
#*# [input_shaper]
#*# shaper_type_x = ei
#*# shaper_freq_x = 88.0
#*# shaper_type_y = mzv
#*# shaper_freq_y = 26.0

```

jschuh, 2023-04-03:
I can't be certain about the first error, but it looks like Creality's Klipper fork removed the [`BED_MESH_CALIBRATE`](https://www.klipper3d.org/Bed_Mesh.html#calibration) gcode. I don't know why they would do that, but it's about the only thing that could cause this error.

The second error is more clear. The Sonic Pad is using a Klipper fork that branched from the mainline version more than six months ago, and thus doesn't include the required support for [`print_stats.info`](https://www.klipper3d.org/Status_Reference.html#print_stats). I originally had code that implemented this functionality, but removed it in 3f1bf6170b12227b5bc8b19bb3862ca3ee31ba44 (a few months after Klipper added [`SET_PRINT_STATS_INFO`](https://www.klipper3d.org/G-Codes.html#set_print_stats_info)).

Unfortunately, this is why I don't support Klipper forks. Too many things can break in unpredictable ways.

closed 2023-04-03 by jschuh

KaratsuFr, 2023-04-04:
@jschuh  thanks for the update
i've fetch the layout before the change and it print well

for the bed mesh : bed mesh calibrate should exist - for now i've update the if statement to force to use your internal command and went ok 

since G29 provided by sonic pad
```
[gcode_macro G29]	
gcode:
  G28
  bed_mesh_calibrate
  G1 X0 Y0 Z10 F4200
```

jschuh, 2023-04-04:
Looks like there was a bug in how the macro was detecting a `BED_MESH_CALIBRATE` override. It should be fixed now with 4945b1b.