jschuh/klipper-macros #64: LOAD_FILAMENT and UNLOAD_FILAMENT not showing up in extruder menu on Mainsail

agreen, 2023-04-07:
Mainsail has a kebab menu if it detects the LOAD_FILAMENT and UNLOAD_FILAMENT macros, however the code in mainsail to display these also check that the gcode within the macro contains a heat wait command ('printer.extruder.can_extrude', 'TEMPERATURE_WAIT', 'M109') - Since the LOAD and UNLOAD macros don't contain any of these commands, the kebab menu with the load/unload options are hidden.

agreen, 2023-04-07:
This almost seems  like a Mainsail bug as macros aren't case sensitive within Klipper.  Opening up a bug within Mainsail as well. Will comment with the bug # on the Mainsail repo.

jschuh, 2023-04-07:
Ah thanks. I verified that it works if I change the case of the command and add an `M109` line to the dummy block that's already there to trigger the parameter hinting. It's a tiny change, so maybe I'll just land it with a TODO explaining the case difference.

There's a growing number of these things where Mainsail/Fluidd implemented some UX functionality that gets enabled when a macro tickles the right behavior. I also saw that they recently added pausing at layer changes [after I suggested it last year](https://github.com/Klipper3d/klipper/pull/5726#issuecomment-1257237360), but I find the implementation significantly lacking compared to the one I already have. So, I was thinking about checking the code out and writing a PR to improve it and make it more compatible with mine. Maybe I'll start with a PR for the casing bug while I'm at it.

closed 2023-04-07 by jschuh

agreen, 2023-04-08:
FYI Looks like this was fixed in the mainsail codebase today, from the bug I opened there. 

They didn't fix the bug I found in the set layer logic though.  I do agree yours seems a bit more advanced. I like that yours works in Klipper. Their implementation seems to rely on Mainsail to do the work of pausing at exactly the right time.

jschuh, 2023-04-08:
Thanks for the heads up. Also, the logic for their layer implementation is actually in their `SET_PRINT_STATS_INFO` override macro, which you can pull from the [mainsail-config repo](https://github.com/mainsail-crew/mainsail-config). So, it will work from the console, but it just provides one command for a scheduled layer and another one for the next layer. Whereas my implementation supports any number of commands at arbitrary layers and heights, either before or after the layer change.

I'm not sure they'd want to expose that level of complexity, but it would be nice to not be limited to just two pending commands. Of course, if I get them to accept the PR I'd have to do it in multiple steps across four repos (to cover both Mainsail and Fluidd) which is why I've procrastinated a bit.