jschuh/klipper-macros #70: [BUG] Errors after updating to the latest repo

b-pardeep, 2023-04-15:
After updating to the latest repo, I'm encouintering 2 errors:

4:35 AM
Must home axis first: 0.000 0.000 2.000 [-1.000]
4:35 AM
Must home axis first: 0.000 0.000 2.000 [-1.000]
4:35 AM
Must home axis first: 0.000 0.000 2.000 [-1.000]
4:35 AM
The value 'resume' is not valid for MACRO
4:35 AM
The value 'resume' is not valid for MACRO
4:35 AM
The value 'resume' is not valid for MACRO

When I try manual homing to get rid of the first error, then I get this error instead:
4:37 AM
Extrude below minimum temp
4:37 AM
Extrude below minimum temp
See the 'min_extrude_temp' config option for details
4:37 AM
Extrude below minimum temp
4:37 AM
Extrude below minimum temp
See the 'min_extrude_temp' config option for details
4:37 AM
Extrude below minimum temp
4:37 AM
Extrude below minimum temp
See the 'min_extrude_temp' config option for details

Then I tried manual setting my temp to 200/60 to get rid of the second error, after that I'm left with only this error.

4:46 AM
The value 'resume' is not valid for MACRO
4:46 AM
The value 'resume' is not valid for MACRO
4:46 AM
The value 'resume' is not valid for MACRO
4:46 AM
The value 'resume' is not valid for MACRO
4:46 AM
The value 'resume' is not valid for MACRO
4:46 AM
The value 'resume' is not valid for MACRO
4:46 AM
The value 'resume' is not valid for MACRO

I tried printing the jobs that I had already printed before just to double check that my slicer isn'y broken or anything but ended up receiving the same errors. I didn't change any settings for my printer/configs other than updating the klipper-macros.

bowuesser8107, 2023-04-15:
Hi, exactly the same for me!
... luckily I still had an older backup of the Macro folder!
Sicherlich nur ein kleiner Fehler der bald behoben ist!
Surely only a small error that is soon fixed! Because I love these automated Klipper-Macros

jschuh, 2023-04-15:
Sorry, I accidentally pushed an incomplete change yesterday and didn't realize that was the case until just now. It's been reverted in 48170bc549d2dbec6566a22cf30f0c1262779150.

closed 2023-04-15 by jschuh

b-pardeep, 2023-04-15:
Thank you for the quick response and even quicker fix.