jschuh/klipper-macros #72: [BUG] Error evaluating 'gcode_macro SET_PAUSE_AT_LAYER:gcode': TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'

ariekraakjr, 2023-04-17:
Since the latest release i got a TypeError on the  SET_PAUSE_AT_LAYER macro.

Klipper: v0.11.0-173-ga8b1b0ef
klipper-macros: v0.5.5-1-gd028f05

jschuh, 2023-04-17:
What slicer are you using, and can you share the startup gcode from your slicer config?

Also, exactly when during the print did you issue this command?  Because, `print_stats` contains `NoneType` [only before it's been initialized](https://github.com/Klipper3d/klipper/blob/b0ffb269d2ac8dc19ae6879d80718b8c60ab60de/klippy/extras/print_stats.py#L74), but it gets initialized at the very beginning of [`PRINT_START`](https://github.com/jschuh/klipper-macros/blob/48170bc549d2dbec6566a22cf30f0c1262779150/start_end.cfg#L35). So, it doesn't seem like there could be a window where the print is active and `print_stats` is not initialized.

Accepting that, I just added the `SET_PAUSE_AT_LAYER` wrapper last week, to better integrate with the Fluidd and Mainsail front-ends. So, it's possible there's some corner case I missed in my testing.

ariekraakjr, 2023-04-18:
I took a look at it today. I'm using the default (yours) start and end g code. The message only appears if you cancel a print.

jschuh, 2023-04-18:
So the `SET_PAUSE_AT_LAYER` was in the gcode command queue after you cancelled the print? In that case it's expected behavior, because the command can't run when there's no active print. I recently made some changes to those macros in 162259645a7e3a299289fe98f2c6ffc76c387430, which should improve the error message when that happens, but it's still an expected error.

Also, if you're going to make a bug report in the future please follow the [instructions from the readme](https://github.com/jschuh/klipper-macros#reporting-bugs). It would have saved me time and avoided the extra questions if the config information and logs had been attached.

closed 2023-04-18 by jschuh