jschuh/klipper-macros #75: Pause or unload_filament pauses leaving the nozzle in place rather than moving it up and out of the way

derekyle, 2023-04-21:
It appears in the config that it should move the head to variable_park_x, y, z. Which don't appear to be set anywhere, so I added them to the _km_options section of the config. but no luck. Is there something else that needs to be configured?

I also tried m702 which is supposed to pause and the unload the filament, but I get the message that the print is not paused. m600 pauses the print but also does not move the head away from the print.

jschuh, 2023-04-21:
The `PAUSE` macro [always parks the printhead](https://github.com/jschuh/klipper-macros/blob/48170bc549d2dbec6566a22cf30f0c1262779150/pause_resume_cancel.cfg#L27), and the `M702` and `M600` commands simply wrap the `PAUSE` macro with some extra functionality. So, it doesn't appear that you're actually calling the macros in this repo.

Usually this sort of report is due to conflicting macro definitions or a similar config error, but I can't be sure since none of the information required for diagnosis was included in the report. The readme provides some [troubleshooting guidance](https://github.com/jschuh/klipper-macros#troubleshooting) and instructions on the necessary information to include in bug reports.

closed 2023-04-21 by jschuh

derekyle, 2023-04-22:
The issue, in case anyone else comes across this, was that I wasn't using the virtual_sdcard component. Or at least, I didn't have my octoprint directory set there. It appears to be required for a lot of these macros to work correctly. I wasn't aware of how the GCODE is handled differently when klipper thinks it is printing from an SD card versus printing from the host.

jschuh, 2023-04-22:
Ah, it hadn't even occurred that you'd be streaming the print via Octoprint, rather than using the virtual_sdcard. I expect could make the `PAUSE` macro still work in this case, but the truth is that streaming gcode to Klipper introduces a range of potential issues, so it's best to just always use the virtual_sdcard for printing.