jschuh/klipper-macros #76: [BUG] Voron TAP issue with Start End Marco

M3HNGRY, 2023-04-23:
Before making a report please ensure you've followed the troubleshooting steps at:
https://github.com/jschuh/klipper-macros#troubleshooting

**Describe the bug**
The "probe too hot, waiting to hit 150 degrees' or whatever message it is is fine, it will just sit there until it has cooled down. The problem people run in to is they don't put an M109 S{printing temperature} after it so it will try to start printing at 150 then error out (or jam)

**Klipper errors**
Extruder temperature target of 240.0C is too high, lowering to 150.0C

**Klipper config**
Printer.cfg

This G-code will allow you to probe cold, but will also prevent you from probing with a nozzle at printing temperature (to try to preserve your build surface). This goes in the [probe] section of your config.

activate_gcode:
    {% set PROBE_TEMP = 150 %}
    {% set MAX_TEMP = PROBE_TEMP + 5 %}
    {% set ACTUAL_TEMP = printer.extruder.temperature %}
    {% set TARGET_TEMP = printer.extruder.target %}

    {% if TARGET_TEMP > PROBE_TEMP %}
        { action_respond_info('Extruder temperature target of %.1fC is too high, lowering to %.1fC' % (TARGET_TEMP, PROBE_TEMP)) }
        M109 S{ PROBE_TEMP }
    {% else %}
        # Temperature target is already low enough, but nozzle may still be too hot.
        {% if ACTUAL_TEMP > MAX_TEMP %}
            { action_respond_info('Extruder temperature %.1fC is still too high, waiting until below %.1fC' % (ACTUAL_TEMP, MAX_TEMP)) }
            TEMPERATURE_WAIT SENSOR=extruder MAXIMUM={ MAX_TEMP }
        {% endif %}
    {% endif %}

closed 2023-04-24 by jschuh

M3HNGRY, 2023-04-24:
Thanks for responding and keeping up with this. It's very useful for me as a new Voron Builder. Anyone who read this ticket and able to help please do. Willing to buy you coffee or beer for your time. TIA

jschuh, 2023-05-01:
Circling back, it seems like all this really needs is a `variable_start_extruder_probe_temp`. I was thinking about doing that anyway, since a few people have asked for it in favor of the current `variable_start_extruder_preheat_scale` behavior. That's easy enough to test without needing TAP hardware, so I'll probably do it this week.

M3HNGRY, 2023-05-01:
Awesome and Thank you! :D

M3HNGRY, 2023-05-10:
To update! New code update works great with tap 👯