jschuh/klipper-macros #87: [FEATURE REQUEST?!] Filament runout sensor support

remigod86, 2023-05-08:
Issue: Because I haven't seen any specific Macro in your collection for runout sensors, I have created my own config (below) from a tutorial.
Whenever the sensor detects an issue the pause_on_runout issues the Parking routine, which seems to work as intended. Trying to resume the print, issues the "Printer is not paused" error. Trying to sop the print, issues the "No active print." Error. The only solution is issuing the Emergency stop.

Can you please help me in understanding the issue or even finding a solution?

Errors when nozzle gets parked:

11:27 PM
Error evaluating 'gcode_macro _km_check_is_printing:gcode': gcode.CommandError: No active print.
11:27 PM
Printer is not paused.

filament_runout_sensor.cfg:
[filament_motion_sensor smart_filament_sensor]
pause_on_runout: True
detection_length: 20.0
event_delay: 3.0
pause_delay: 0.5
extruder: extruder
switch_pin: !PG12 # Pin that the sensor is connected to
runout_gcode:
  G91
  G1 Z10 F900      # Raise Z away from print
  G90
  G1 X10 Y10 F1200 # Move to purge area
  G91
  G1 E-20 F600 # Retract 20mm of filament
  G90
insert_gcode:
  G4 P10000 # Pause for 10 seconds
  G1 E150 F400 # Prime the nozzle

# [delayed_gcode DISABLEFILAMENTSENSOR] ; This will disable the SFS 1 second after klipper starts
# initial_duration: 1
# gcode:
#     SET_FILAMENT_SENSOR SENSOR=smart_filament_sensor ENABLE=0 ; Put your filament sensor's name after SENSOR=

[gcode_macro SFS_ENABLE] ; Add this to PRINT_START
description: Enable smart filament sensor
gcode:
    M117 ENABLING the Smart Filament Sensor
    G92 E0
    SET_FILAMENT_SENSOR SENSOR=smart_filament_sensor ENABLE=1 ; Put your filament sensor's name after SENSOR=

[gcode_macro SFS_DISABLE] ; Add this to PRINT_END and PRINT_CANCEL
description: Disable smart filament sensor 
gcode:
    M117 DISABLING the Smart Filament Sensor
    G92 E0
    SET_FILAMENT_SENSOR SENSOR=smart_filament_sensor ENABLE=0 ; Put your filament sensor's name after SENSOR=

jschuh, 2023-05-09:
You might want to ask in issue #86, since that person appears to have a working config, and I don't use any `filament_motion_sensor` hardware.

More generally, I'm trying to avoid the added complexity of dealing with non-macro configs in this project (which is why I don't have any runout configs here). I do use `filament_switch_sensor` hardware on my own printers, but this is all I do to configure them:

```
[filament_switch_sensor runout]
pause_on_runout: False
switch_pin: <replace with pin>
runout_gcode:
  {% if printer.print_stats.info.current_layer <=
          printer.print_stats.info.total_layer and
        not printer.pause_resume.is_paused %}
    PAUSE_NEXT_LAYER
  {% endif %}
```

Also, if you make bug reports in the future please be sure to use the code block formatting for any cnofig snippets, otherwise they're impossible to read.

closed 2023-05-09 by jschuh

mjenda, 2023-05-09:
Good timing, yesterday I run into the same issue. In Klipper repo I found that when runout (or smart) sensor handles the runout event it:
1. pauses the printer by calling `send_pause_command` on `[pause_resume]` ([link](https://github.com/Klipper3d/klipper/blob/3c79e484d5905f97a2376408830f33631b8ca36c/klippy/extras/filament_switch_sensor.py#L51)) which eventually exits SD card from printing 
2.  and **then**, one line later it executes `PAUSE` code 

As a result, condition `{% if printer.virtual_sdcard.is_active %}` is false in `pause_resume_cancel.cfg` ([link](https://github.com/jschuh/klipper-macros/blob/main/pause_resume_cancel.cfg#L17)).

In first link I pasted you can see that `send_pause_command` method is being called only if `pause_on_runout` is set to `true` in config. So in my setting I fixed this by setting it to `false` and executing `pause` command in `runout_gcode`.
```
[filament_motion_sensor sensor_name]
...
pause_on_runout: False
runout_gcode:
    PAUSE
```

IMO this suggestion can be added somewhere to documentation in order to prevent other requests or bugs reports.

This "issue" can also be fixed by changing the condition `pause_resume_cancel.cfg` to something different. For example in [Voron documentation](https://docs.vorondesign.com/community/howto/samwiseg0/btt_smart_filament_sensor.html) I found that they suggest using`{% if printer['pause_resume'].is_paused|int == 0 %}` in pause macro.

jschuh, 2023-05-09:
Yeah, I mentioned in #88 that I'm going to make the print/pause status checking a bit less strict. I'll do that in the next few days.

jschuh, 2023-05-09:
Okay, I'll probably merge dev later today, which will pull in 0f3e3257f87112f3957f812b8b7c4d9f9d3bba8d to address the pause thing.

mjenda, 2023-05-09:
Thanks for this fix, it looks good to me. For the double check I switched the branch and the printer pauses flawlessly with default runout sensor config (with `pause_on_runout: True` ).