jschuh/klipper-macros #89: [FEATURE REQUEST]

fire1ce, 2023-05-09:
Hey @jschuh 

It's not my first time opening an "issue" at your Git repository.
I'm using your work daily, and it's amazing!
The only way to communicate right now in this Git repo is by opening issues (like I'm doing right now), but I think it's not the best, since most of the "issues" being opened here are just "help" topics or suggestions. When they are closed, the knowledge of that topic is "lost" in the closed issues section.

May I suggest enabling Discussions for this repo? It may allow users like me to open topics and get help not only from you, but also from others. I know it can be daunting to "resolve" the issues that are being opened.
This would allow other users to discuss and suggest solutions, workarounds, and even get ideas from those topics.
You could also update your README to link to the Discussions, rather than prompting users to open an issue for every question.

[image]

jschuh, 2023-05-09:
Fair enough. I'll take a look at it this week and see about enabling it.

M3HNGRY, 2023-05-10:
Agreed this would be so helpful! Great suggestion.

jschuh, 2023-05-10:
Poked at it a bit earlier today and accidentally enabled it. So, I just updated the docs and issue template to point to it, and made an initial post.

closed 2023-05-10 by jschuh