jschuh/klipper-macros #90: [FEATURE REQUEST] Ability to keep the printer preheated after a print

agreen, 2023-05-09:
I'm printing my next Voron, and I'm batch printing a lot of parts on my v0.  It'd be helpful to keep the bed (and chamber) preheated between prints.

My process typically goes, print finishes - remove hot spring steel from bed, and place on a cooling location, close printer door and leave preheating.  Print cools, remove print, clean bed, reinstall and I'm ready to start next print.

My suggestion - Have a variable that can be set/unset via  macro to enable/disable the feature.  If the variable is set, the bed will NOT cool down after the print, but stay heating.  This could additionally change the idle timeout and/or keep the nozzle preheated as well (I have a preheat profile right now which keeps the nozzle at half my print temp, and the bed at full temp, to keep the chamber temp up hot enough for ABS).

jschuh, 2023-05-11:
Now that there's [a section for this sort of thing](https://github.com/jschuh/klipper-macros/discussions/categories/ideas) in the discussions tab I think it would be better to make a post there.

That stated, I don't think I'm inclined to do what's described because it doesn't seem like something that makes sense to persist as a config option rather than through a slicer profile. So, for this particular slicer profile you'd add your code for keeping things warm to the end g-code section, after the PRINT_END line. That way whenever you print with this profile it would always keep everything warm after completing.

agreen, 2023-05-11:
Didn't think of the obvious here.. just making a new printer profile to leave it on... Missed the obvious.