jschuh/klipper-macros #93: [BUG] Bed mesh violates bed limits

agreen, 2023-05-10:
**Describe the bug**
Please provide a clear and concise description of the issue.
I have a large print on the bed and am using the fast bed mesh. My start gcode contains the mesh_min and mesh_max variables.  Additionally, I'm using a Klicky probe, which included in their config is to set the relative reference index.

Reviewing bed_mesh_fast, the majority of the logic is bypassed, however all params passed in are passed out to the native bed_mesh_calibrate (since as stated in the code, not really sure what/how to handle the RRI).

The issue is, PASSING the mesh_min and mesh_max seem to override Klipper's mesh_min and mesh_max that were configured, and you can violate your min and max that were configured in the config file (and get a move out of range error).

agreen, 2023-05-10:
Two suggested approaches - DROP the mesh_min and mesh_max parameters if you are bypassing the the custom logic, or clamp the mesh_min and mesh_max to the safe coords regardless of what to do with the RRI.  I would suggest the first option. It would seem my bed ALWAYS meshes the entire thing anyway, no matter what I put on it, but for some reason when mesh_min is smaller than the configured, it accepts it and creates a move out of range.

I recognize by the way, this is probably a bit more of a Klipper problem than a macro problem, but since this macro is allowing you to violate what's been configured..... /shrug.. not really sure where it should be fixed

jschuh, 2023-05-10:
I'll just dump the extra params when it falls back. While I'm at it I might as well handle RELATIVE_REFERENCE_INDEX by just grabbing the middle point in the grid and passing it to BED_MESH_LEVEL.

If I'm feeling really ambitious I suppose I could make MESH_RADIUS work, and drag the delta out to test it. Then I could get rid of the fallback. Although, I doubt I'm feeling that ambitious.

jschuh, 2023-05-10:
Considering it a bit more, I think dropping the mesh parameters on the fallback path is a bad idea. It seems that it would only mask an issue where the slicer could generate print moves out of bounds. Better to check the bounds on both paths and error out early.

agreen, 2023-05-10:
if I had a delta I'd help you test and/or help fix the logic there...  

You could provide an idea of where to put the relative_reference_index - which corner/midpoint, if you felt REALLY ambitious.... :D

jschuh, 2023-05-11:
I made the checking stricter in a5918abb8980d522948c4ea39f7f4d141e1411f0, and now it will warn if the config allows a bad mesh and abort if a bad mesh would be generated.

I also made RELATIVE_REFERENCE_INDEX work in 99bc290be81288c5006754ab4069f2e2b19c2f21 by finding the closest index point in the optimized mesh to the original index point in the mesh the unmodified config would generate.

That stated, the core issue you were hitting really seems to be that the slicer is outputting mesh params that are outside the printable bounds. So, it seems like that really needs to be fixed in the slicer config.

closed 2023-05-11 by jschuh

agreen, 2023-05-11:
I don't think this is a slicer config issue.  As an example, here is my mesh setting from a recent print that has this issue.  MESH_MIN=7.98606,7.98606 MESH_MAX=112.676,109.032

The issue is, my MESH_MIN is 15,15 in the printer.cfg - and needs to be, due to my probe offset.  There is still that extra 15mm I can use to PRINT with though, but due to the probe offset, I can't probe that area.  According to the Klipper docs here https://www.klipper3d.org/Bed_Mesh.html 

> mesh_min: 35, 6
> Required
> The first probed coordinate, nearest to the origin. This coordinate is relative to the probe's location.

It's the fact that this isn't the NOZZLE position, but the PROBE position that creates the possibility for the nozzle to have invalid moves.

jschuh, 2023-05-11:
Ah, yeah I somehow paged that part out of my brain when I was doing the RRF part. I guess since deltas are the only thing left in the fallback path I'll just strip the arguments. It wouldn't be hard to just check that they fall in the mesh_radius, but I'd rather just do that at the same time I do the rest of the delta support.