jschuh/klipper-macros #100: Pause macro does not work correctly

DarkPako84, 2023-05-13:
After the last update, immediately after pressing the resume command, the hotend does not reposition correctly and extrudes into the air. It also does not resume the correct speed but goes much slower.

jschuh, 2023-05-13:
I can't think of any recent changes to part of the macros, and I can't debug an issue without more context. So, please follow the [troubleshooting and bug reporting guidance](https://github.com/jschuh/klipper-macros#troubleshooting), and I'll reopen if this if there's more detail that identifies an issue.

hardmax, 2023-05-14:
Hi,

Same problem for Here. After M601, pause occour (Park with beep) after that the nozzle go over print in correct position., but appear the printer do an extra park, again, noozle go over print but not in layer and resume print but with z in other layer.

I think that in my config are a second pause and resume, and the second one last the actual layer heigh.

I hope that this help

Regards

jschuh, 2023-05-14:
Looking a bit more closely, I'm certain that a7a34ab1604643a591ed4af62fb7188c800a781e is just wrong, so I already reverted it from main. You should be able to just update as normal and I expect everything will be fine.

closed 2023-05-14 by jschuh

hardmax, 2023-05-15:
Yes, now all are ok, are working...thanks for you work...

A new feature that I like propuse you is add a varible to end print macro to increase last retract. This is because the new easy change nozzle like E3D Revo and Rapido and others need that at finish print with a long retract releases the filament from the nozzle when the print finishes. As a result, the nozzle can be unscrewed without heating up and removing the filament first.  

Thanks againg