jschuh/klipper-macros #97: Chamber heater not heating

ValSimDim, 2023-05-13:
Hi @jschuh, just installed on my SK Cube this wonderful set of macros you have created. I'm very new to macros and still need to understand a lot of things. I'm using Orca slicer, Prusa slicer clone, which works very well for me. Managed to get the printer printing with all bed mesh, tramming, temps, start and end gcodes, all works fine, except the Chamber heater doesn't start heating to the required temperature at the beginning of the print. I can set the chamber temperature manually and I have setup fan which is turning on and off when the chamber heater is working and all this works fine manually, but not working automatically when I start a print. I did set up chamber temperature in the slicer and included " CHAMBER=[chamber_temperature] " in the start gcode of the slicer printer config.

I'll appreciate if you can take a look at my configs below and direct me what I'm doing wrong:

Slicer start_gcode:

""

M190 S0
M109 S0
PRINT_START EXTRUDER={first_layer_temperature[initial_tool]} BED=[first_layer_bed_temperature] MESH_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} MESH_MAX={first_layer_print_max[0]},{first_layer_print_max[1]} LAYERS={total_layer_count} NOZZLE_SIZE={nozzle_diameter[0]} CHAMBER=[chamber_temperature]

""

printer.cfg heater_generic chamber and temperature_fan chamber_fan section:

""

[heater_generic chamber]
gcode_id: chamber
##   The id to use when reporting the temperature in the M105 command.
##   This parameter must be provided.
heater_pin: PA3    # HE1
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PF5		# T1
#control: pid
control: watermark
#pid_Kp: 24.921
#pid_Ki: 1.093
#pid_Kd: 142.050
## max_power ranges between 0.0 ~ 1.0. 
## Lower this if the thermal switch piggy-backed on 
## the PTC heater activates frequently.
max_power: 0.3		
min_temp: 0
max_temp: 70
max_delta: 2.0
pwm_cycle_time: 0.0166

[verify_heater chamber]
max_error: 900
check_gain_time: 900
hysteresis: 3
heating_gain: 1

[temperature_fan chamber_fan]
pin: PD12
max_power: 1.0
shutdown_speed: 1
kick_start_time: 0.5
sensor_type: Generic 3950
sensor_pin: PF6
control: watermark
max_delta: 2.0
min_temp: 10
max_temp: 150
target_temp: 40.0

""

I tried to troubleshoot and actually manage to start the heater, after replacing just for test your start_end macro  section :

""
{% set CHAMBER = params.CHAMBER|default(0)|float
                   if "chamber" in printer.heaters.available_heaters else 0.0 %}

""

to include: default(params.CHAMBER_TEMP) instead of default(0):

""
{% set CHAMBER = params.CHAMBER|default(params.CHAMBER_TEMP)|float %}

""

It starts the chamber heater, but then Klipper is throwing the following error and canceling the print after bed level and tramming and before start printing:

Error evaluating 'gcode_macro _km_park_if_needed:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'chamber'

I also tried:

"" 
{% set CHAMBER = params.CHAMBER|default(params.CHAMBER_TEMP)|float
                   if "chamber" in printer.heaters.available_heaters else 0.0 %}

""

Bu with this the heater temp is not setand the print just starts without the chamber heater heating.  This made me think that the problem is that "chamber" is not in "printer.heaters.available_heaters" but I have no idea what this is. I then reverted the start_end.cfg to its original state, to avoid completely messing up your wonderful job :-)

I'll appreciate any advice to resolve this since I build this printer specifically for ABS and PC printing and automatic chamber heater control would be great.

jschuh, 2023-05-13:
Hm... I haven't actually had a chamber heater set up for years on any of my printers, and the `START_PRINT` has changed quite a bit in that time (also, some of the underlying Klipper code has changed in the interim). So, I'm not surprised that I broke it.  I'll take a look today or tomorrow and try to get it working again.

I have two requests:

1. If I point you to a custom branch, would you be able to test it on your printer?
2. Would you mind wrapping the config snippets in your comments with [markdown code blocks](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)? It's just really hard to read otherwise,

ValSimDim, 2023-05-13:
Thanks for your reply, Justin!

1. I will gladly test whatever you request, no problem at all, just give me clear instructions, not very familiar with the terminology.
2. I would gladly do that, if you give me an example, no idea what "wrapping the config snippets" and "markdown code blocks" mean, I will read the link and do my best!

ValSimDim, 2023-05-13:
Let me try with my printer.cfg section below

`
[heater_generic chamber]
gcode_id: chamber
##   The id to use when reporting the temperature in the M105 command.
##   This parameter must be provided.
heater_pin: PA3    # HE1
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PF5		# T1
#control: pid
control: watermark
#pid_Kp: 24.921
#pid_Ki: 1.093
#pid_Kd: 142.050
## max_power ranges between 0.0 ~ 1.0. 
## Lower this if the thermal switch piggy-backed on 
## the PTC heater activates frequently.
max_power: 0.3		
min_temp: 0
max_temp: 70
max_delta: 2.0
pwm_cycle_time: 0.0166

[verify_heater chamber]
max_error: 900
check_gain_time: 900
hysteresis: 3
heating_gain: 1

[temperature_fan chamber_fan]
pin: PD12
max_power: 1.0
shutdown_speed: 1
kick_start_time: 0.5
sensor_type: Generic 3950
sensor_pin: PF6
control: watermark
max_delta: 2.0
min_temp: 10
max_temp: 150
target_temp: 40.0`

ValSimDim, 2023-05-13:
`
[heater_generic chamber]
gcode_id: chamber
##The id to use when reporting the temperature in the M105 command.
##This parameter must be provided.
heater_pin: PA3    # HE1
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PF5		# T1
#control: pid
control: watermark
#pid_Kp: 24.921
#pid_Ki: 1.093
#pid_Kd: 142.050
##max_power ranges between 0.0 ~ 1.0. 
##Lower this if the thermal switch piggy-backed on 
##the PTC heater activates frequently.
max_power: 0.3		
min_temp: 0
max_temp: 70
max_delta: 2.0
pwm_cycle_time: 0.0166

[verify_heater chamber]
max_error: 900
check_gain_time: 900
hysteresis: 3
heating_gain: 1

[temperature_fan chamber_fan]
pin: PD12
max_power: 1.0
shutdown_speed: 1
kick_start_time: 0.5
sensor_type: Generic 3950
sensor_pin: PF6
control: watermark
max_delta: 2.0
min_temp: 10
max_temp: 150
target_temp: 40.0`

ValSimDim, 2023-05-13:
The output of   listvars s=available_heaters :

`printer.heaters.available_heaters : ['heater_bed', 'heater_generic chamber', 'extruder']`

jschuh, 2023-05-13:
Thanks, the formatting helps. I'm pretty sure I know what I broke (and it's probably been broken for a few years).

I just pushed [a branch with what I think will fix it](https://github.com/jschuh/klipper-macros/compare/main...heaters). 

You'll need to SSH into your printer host and run the following commands (I'm assuming you checked out the code to `~/klipper_config/klipper-macros`):

```
cd ~/klipper_config/klipper-macros
git reset --hard HEAD
git fetch
git checkout heaters
```

Then just restart Klipper and see if everything works.

ValSimDim, 2023-05-13:
Unfortunately after running the commands,  restarting the host and trying to print the same file, the chamber heater temp was not set. It just did as before. Not sure if I did everything correct, see below copy/paste from the host:

`valld@SKCUBE:~ $ cd klipper-macros
valld@SKCUBE:~/klipper-macros $ git reset --hard HEAD
HEAD is now at 24d5be2 bed_mesh_fast: Fix limits on fallback path
valld@SKCUBE:~/klipper-macros $ git fetch
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), 373 bytes | 31.00 KiB/s, done.
From https://github.com/jschuh/klipper-macros
 * [new branch]      heaters    -> origin/heaters
valld@SKCUBE:~/klipper-macros $ git checkout heaters
Branch 'heaters' set up to track remote branch 'heaters' from 'origin'.
Switched to a new branch 'heaters'
valld@SKCUBE:~/klipper-macros $ `

ValSimDim, 2023-05-13:
No idea why the text is showing like this I used the CTRL+E to comment it

ValSimDim, 2023-05-13:
Just noticed you said restart Klipper, instead I restarted the Raspberry, not sure if this is a problem

ValSimDim, 2023-05-13:
I compared my start_end.cfg file with the commit in the link that you posted above and it looks like my file didn't change, did I do something wrong? Below is the section from the current start_end.cfg:

`[gcode_macro print_start]
description: Inserted by slicer at start of print.
  Usage: PRINT_START BED=<temp> EXTRUDER=<temp> [CHAMBER=<temp>]
                     [MESH_MIN=<x,y>] [MESH_MAX=<x,y>] [LAYERS=<num>]
                     [NOZZLE_SIZE=<mm>]
gcode:
  CHECK_KM_CONFIG # Need this in case startup errors were missed.

  M107 # Turn off part cooling fan in case it was on.
  SET_GCODE_VARIABLE MACRO=print_end VARIABLE=was_cancelled VALUE="{False}"
  CLEAR_PAUSE
  {% set BED = params.BED|default(params.BED_TEMP)|float %}
  {% set EXTRUDER = params.EXTRUDER|default(params.EXTRUDER_TEMP)|float %}
  {% set CHAMBER = params.CHAMBER|default(0)|float
                   if "chamber" in printer.heaters.available_heaters else 0.0 %}
  {% set settings = printer["gcode_macro print_start_set"].settings %}
  {% set MESH_MIN = params.MESH_MIN|default(settings.MESH_MIN)|default(None) %}
  {% set MESH_MAX = params.MESH_MAX|default(settings.MESH_MAX)|default(None) %}
  # Clear the print settings after they are no longer needed.
  SET_GCODE_VARIABLE MACRO=print_start_set VARIABLE=settings VALUE="{{}}"
  {% set LAYERS = params.LAYERS|default(settings.LAYERS)|default(0)|int %}
  {% set NOZZLE_SIZE = params.NOZZLE_SIZE|default(settings.NOZZLE_SIZE)|
         default(printer.configfile.settings.extruder.nozzle_diameter)|float %}
  {% set km = printer["gcode_macro _km_globals"] %}
  {% set actions_at_temp = km.start_level_bed_at_temp or
                           km.start_quad_gantry_level_at_temp or
                           km.start_z_tilt_adjust_at_temp %}
  # The bed started at no more than 0.2C below and 1.0C above the target temp.
  {% set bed_at_target = (BED + 0.4 - printer.heater_bed.temperature) |
                           abs <= 0.6 %}
  {% set bed_overshoot = (BED + (km.start_bed_heat_overshoot if
             (BED and not bed_at_target) else 0.0),
           printer.configfile.settings.heater_bed.max_temp ) | min %}`

ValSimDim, 2023-05-13:
Shall I manually edit the start_end.cfg file and test again?

jschuh, 2023-05-13:
I'm testing locally with the chamber mocked out, and it's definitely generating the commands with that patch. My best guess is that when moonraker restarted it reset the `klipper-macros` checkout to whatever you have configured in the moonraker autoupdate section.

So, let's try it again. SSH into your printer host and run the following commands in the `klipper-macros` directory:

```
git reset --hard HEAD
git fetch
git checkout heaters
```

Then run this:

```
git status
```

The first line of output should read `On branch heaters`

Then restart **only Klipper** and run this command in the Klipper console:

```
PRINT_START BED=50 EXTRUDER=180 CHAMBER=40
```

You should see the chamber start heating.

Regarding formatting: single ticks are for inline code blocks. Multi-line code blocks need three ticks on their own line (I edited your last few comments to fix it): https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

ValSimDim, 2023-05-13:
Sorry, my bad, I followed blindly your `~/klipper_config/klipper-macros` and ended up in a wrong folder.... The correct folder is printer_data/config/klipper-macros. I ran everything and the heater chamber now works beautifully!

 Thank you so much Justin, for your time and devotion! I will learn your macros extensively and probably ask questions, I guess the Q&A is the correct place for this, right?

Have a great evening!

jschuh, 2023-05-13:
Glad we got it sorted, and yes, please post config questions to the discussion section.

I just pushed the fix to main, so you'll want to cd into your `klipper-macros` directory and run the following commands to get everything back to normal:

```
git checkout main
git pull
git branch -d heaters
```

The last command might give you a warning about deleting "deleting branch 'heaters'" but don't worry about it, since you pulled in the patch with the new main branch.

closed 2023-05-13 by jschuh