jschuh/klipper-macros #101: [Feature Request]

nitroman123, 2023-05-14:
I would love to have a buzzer sound when the filament change has come or the print is finished m300 doesnt work...

jschuh, 2023-05-14:
Feature requests belong in the [Ideas discussion section](https://github.com/jschuh/klipper-macros/discussions/categories/ideas).

That stated, assuming your printer board has a hardware buzzer/beeper, then the m300 command is available if you [set up the output pin as mentioned in the readme](https://github.com/jschuh/klipper-macros#beep).

closed 2023-05-14 by jschuh

nitroman123, 2023-05-14:
i use sonic pad and as far is i know m300 is not supported throw klipper , i am wrong ?

jschuh, 2023-05-14:
The previous comment links to the documentation for the `M300` implementation in these macros. The implementation itself is in [this file](https://github.com/jschuh/klipper-macros/blob/main/beep.cfg). But as the documentation states, you need to have properly configured the output pin (and the printer board must actually have a beeper).

nitroman123, 2023-05-15:
thanks for that , will it work for sonic pad?