jschuh/klipper-macros #103: [BUG?] Move exceeds maximum extrusion error after resume printing

Yveaux, 2023-05-15:
When pausing a print (using Mainsail's pause-print button), followed by resuming (also through Mainsail's play-print button) the printer consistently comes with the following error:

> Move exceeds maximum extrusion (2.338mm^2 vs 0.640mm^2)
> See the 'max_extrude_cross_section' config option for details

No other commands or macro's are executed between pause and resume.

Defining the `max_extrude_cross_section` value to something big will indeed suppress the error, but I'm trying to understand why this error even occurs. Googling indicates it is caused by a difference in extruder position, which klipper will try to make up for when resuming the print (ref. https://klipper.discourse.group/t/problem-move-exceeds-maximum-extrusion/3941/6).

Could it be that a small extruder positional error is introduced by the macro's?

jschuh, 2023-05-15:
My guess is it was a bug in the `PAUSE` macro that was introduced last week and fixed yesterday. But I can only guess because this report doesn't include the necessary information to diagnose an issue. Next time please follow the [troubleshooting and bug reporting guidance](https://github.com/jschuh/klipper-macros#troubleshooting) from the readme.

Yveaux, 2023-05-16:
Next time I will include the requested configs.

Just to report back, updating to the latest version (v0.7.5) fixed the "Move exceeds maximum extrusion" errors on pause/resume.