jschuh/klipper-macros #109: [BUG] Printing with Octoprint returns "No active print" in PRINT_END macro

lettore, 2023-05-19:
I'm not totally sure if printing with Octoprint is supported, but when the print ends I get "No active print" error and the printer stops at the beginning of PRINT_END macro.
Loading the same file from the virtual_sd returns no errors and the print is completed correctly.
For the moment I print with `SET_GCODE_VARIABLE MACRO=_km_check_is_printing VARIABLE=debug_state VALUE="{True}"` in my printer.cfg so it skips the macro `_km_check_is_printing`.

lettore, 2023-05-19:
The issue resides in commit 0f3e325 
Without this part with Octoprint PRINT_END is working as expected
` not (printer.virtual_sdcard|default({})).is_active|default(False))`

jschuh, 2023-05-19:
Oh, yeah, looks like that line should have been an "and" not an "or".

lettore, 2023-05-20:
Confirm that the correction is working

closed 2023-05-20 by lettore