jschuh/klipper-macros #113: [BUG]

ishampadron, 2023-05-22:
Before making a report please ensure you've followed the troubleshooting steps at:
https://github.com/jschuh/klipper-macros#troubleshooting

You can also get general troubleshooting assistance in the Q&A Discussion tab:
https://github.com/jschuh/klipper-macros/discussions/categories/q-a

**Describe the bug**
I dont have a heated bed configured. 

Klipper reports: ERROR

Existing command 'M190' not found in gcode_macro rename

**Klipper errors**
Paste the full text of any commands that trigger the error, along with any Klipper error messages and relevant portions of the Klipper logs.

**Klipper config**
Paste any relevant sections from your Klipper config, or attach the full config if you prefer. Please also note any other macros you are including as part of your config. **Bug reports without attached configs will most likely be closed.**

jschuh, 2023-05-22:
This is working as intended, and is [called out explicitly in the readme](https://github.com/jschuh/klipper-macros#a-few-warnings).

closed 2023-05-22 by jschuh

rcarmo, 2023-11-05:
Is there no way whatsoever to use your macros without a heated bed? I am trying to get a consistent setup across multiple printers and slicers, but my eldest 0.6mm nozzle printer (which benefits the most from Klipper) doesn't have (and never will have) a heated bed.