jschuh/klipper-macros #119: [BUG] M300 beep macro uses invalid value when using non-pwm beeper

pwNBait, 2023-05-26:
Before making a report please ensure you've followed the troubleshooting steps at:
https://github.com/jschuh/klipper-macros#troubleshooting

You can also get general troubleshooting assistance in the Q&A Discussion tab:
https://github.com/jschuh/klipper-macros/discussions/categories/q-a

**Describe the bug**
[image]
The M300 macro uses an invalid value when using a non-pwm beeper. 
Since a non-pwm beeper cannot have a scale setting, the macro uses a default value of 1.0.
` settings["output_pin beeper"].scale|default(1.0) * 0.5`
However, the default 1.0 * 0.5 results in an invalid value for the digital SET_PIN command.
It should be 1.0 or 0.0. 

**Klipper config**
I am using an Ender 3 v2 Neo

[output_pin beeper]
pin: PB13

pwNBait, 2023-05-26:
That makes sense. Thanks for taking a look.

pwNBait, 2023-05-26:
Sure. My printer is busy at the moment. I'll be able to test the change tommorrow.
I'll let you know if the change works.

pwNBait, 2023-05-27:
I tested the change and it works perfectly.
Let me know if there is anything else to test.

jschuh, 2023-05-27:
Thanks, that's all I needed. I'll land it on main soon.

closed 2023-05-27 by jschuh