jschuh/klipper-macros #140: bed is crashing into Y max position

cliffiard, 2023-06-24:
I have recently installed this on an ender 3 pro using klipper,
The problem that I have is that when printer is parking or it's finishing the print, at print show off, the bed is crashing into the Y axis max position. 

I went through start_end.cfg since I assume this is the place, but honestly I can't figure out what to modify. 

I have reduced the position_max on Y and X, but the problem persisted, and I lose print space as well. 

Could you please direct me to the right place?

closed 2023-06-24 by cliffiard

lsljunior, 2024-05-03:
Hello!
I'm having exactly the same issue, but my printer is a neptune 3 pro.
What exactly did you change in printer.cfg?
Thanks in advance

Trei-D, 2024-05-03:
I have reduced the bed size in printer.cfg. Also I had a belt tensioner
which was taking space.