jschuh/klipper-macros #148: [BUG]Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'

Johnn201, 2023-07-05:
Before making a report please ensure you've followed the troubleshooting steps at:
https://github.com/jschuh/klipper-macros#troubleshooting

You can also get general troubleshooting assistance in the Q&A Discussion tab:
https://github.com/jschuh/klipper-macros/discussions/categories/q-a

**Describe the bug**
When attempting to print a file sliced in superslicer I get the noted error, after printing successfully for 4 days straight this error happened out of the blue.

**Klipper errors**
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'

**Klipper config**
Paste any relevant sections from your Klipper config, or attach the full config if you prefer. Please also note any other macros you are including as part of your config. **Bug reports without attached configs will most likely be closed.**
[klippy (1).log](https://github.com/jschuh/klipper-macros/files/11955879/klippy.1.log)

Johnn201, 2023-07-05:
[start.txt](https://github.com/jschuh/klipper-macros/files/11960443/start.txt)

jschuh, 2023-07-06:
You're running an old version of the macros. The current version handles SuperSlicer's weird EXCLUDE_OBJECT entries. (This is also why the [troubleshooting steps](https://github.com/jschuh/klipper-macros#troubleshooting) include updating to the latest version.)