jschuh/klipper-macros #158: [BUG]  All surface offsets are changed after PROBE_CALIBRATE

lettore, 2023-07-23:
When I do PROBE_CALIBRATE and restart I found that the difference between before and after is added at the z-offset of the bed surfaces. While this is maybe by design when you have a z-endstop and a probe to measure the bed mesh, it's not logical when you have a printer with only the probe sensor. Because with this behaviour it's impossible to fix the z-offset of the nozzle and the probe. I can't say how much time I spent before figuring out the problem. I also don't understand what trigger this action, so I can just simply disable it. At the moment I have to manually correct every surface offset manually.

SarcasmRules, 2023-07-31:
The above is correct, there's a G28 after probing/leveling needed, before extruder stage 2 starts.
When pre-level Z=0 and bed/gantry moves, 0 != 0

SarcasmRules, 2023-08-01:
@lettore 

I've successfully added a G28 Z, however, heating the extruder is already called and may delay the Z homing till it cools off but for now I guess it should work. If you're using the latest (with phased print_start) add this in your custom gcode (slicer print_start):

`_PRINT_START_PHASE_PROBING    
; Insert custom gcode here    

G28 Z    
_PRINT_START_PHASE_EXTRUDER`

closed 2023-08-01 by jschuh