jschuh/klipper-macros #180: [Enhancement] Support multi-level print_end

dpeart, 2023-09-25:
At the end of a print I call a macro clean_nozzle.

I would like to include it as part of the print_end macro, similar to how you support multi-steps in print_start. Today I do a print_end, then clean_nozzle, but it ends up parking the head twice, etc.  Lots of extraneous moving. 

I don't need direct support, but something that can be inserted after retracting the head, but before it is in its final parked location.

Hopefuly this makes sense.

The macro is essentially:
`# CLEAN_NOZZLE: Cleans the nozzle by moving it over the brush

[gcode_macro CLEAN_NOZZLE] 
variable_start_x: 310
variable_start_y: 350
variable_start_z: 11.0 
variable_wipe_dist: -50 
variable_wipe_qty: 10
variable_wipe_spd: 200 
variable_raise_distance: 10 

gcode: 
  PARK P=0 X={start_x} Y={start_y} Z={start_z}

  ## Wipe nozzle 
  {% for wipes in range(1, (wipe_qty + 1)) %} 
    G1 X{start_x + wipe_dist} F{wipe_spd * 60} 
    G1 X{start_x} F{wipe_spd * 60} 
  {% endfor %} ## Raise nozzle 
  G91                            ; relative positioning
  G1 Z{raise_distance}`

HitLuca, 2023-10-30:
Why not putting the call to your macro at the end of `klipper-macros/start_end.cfg:_print_end_inner`?

dpeart, 2023-10-30:
That is what I did.  I guess I should close this.

closed 2023-10-30 by dpeart