jschuh/klipper-macros #182: Help: M600 Filament change, pooping in the middle of print after filament change and clicking resume.

anuragdeshpande, 2023-09-26:
Hello,

First of all, Thank you so much for this great repository. Saved my skin a ton of times than I can mention. 

Issue: 
Not sure if this helps, but, I am using ender 3 s1 pro. I probably may not have noticed or it started happening recently, I am not sure. But, lately when I am pausing the print with M600 Command, the print is pausing, retracting as expected. However, it is the "resuming" the print, where I am running into issues. It poops a little, probably at the last saved location before pausing and then moves on to another location to continue printing. 

I tried reading the macro code for Resume, and it appears as if it is the un-retract block of the code that is pushing a little too much? Here is a picture of when it happens. Please let me know if this is something you could help me configure so that it resumes as beautifully as it pauses. If it helps, I can reprint another piece and record the video when it happens? Any help is appreciated.

Thank you.
[image]

anuragdeshpande, 2023-09-26:
I am assuming this is important piece to the puzzle? I have used the "play/resume" button on the mainsail interface. I got the same result. I am also a user of Mobileraker app for klipper/mainsail. I have used the resume button in that app as well. Both seem to be doing the same thing.

HitLuca, 2023-10-30:
Can confirm that this happens to me as well, the nozzle returns to the last position before pausing, and then moves to the new (proper) location. I don't think it's a fault of this project tho

appleimperio, 2023-11-07:
I am having the same issue did any of you found the solution? Thank you

jschuh, 2023-11-07:
This is just how Klipper's [resume](https://www.klipper3d.org/G-Codes.html#resume) works.

closed 2023-11-07 by jschuh