jschuh/klipper-macros #185: [BUG] Errors in console from default macros

josurr, 2023-10-10:
**Describe the bug**
Receiving the following error in the Klipper console upon printer boot & connect:

`Error evaluating 'gcode_macro check_km_config:gcode': gcode.CommandError: load_length exceeds max_extrude_only_distance.`

The following error occurs numerous times when the printer returns to idle after performing a command (e.g. G28):

`Error evaluating 'gcode_macro _km_print_status:gcode': gcode.CommandError: Status 'none' not valid.`

**Klipper errors**
Attached klippy.log

**Klipper config**
Attached printer.cfg (renamed to .txt because of GitHub restrictions)
[klippy.log](https://github.com/jschuh/klipper-macros/files/12861036/klippy.log)
[printer.txt](https://github.com/jschuh/klipper-macros/files/12861038/printer.txt)

Perrotn, 2023-10-17:
in the [extruder] section
max_extrude_only_distance default is 50mm, in globals.cfg: variable_load_length: 90.0 therefore variable_load_length>max_extrude_only_distance 

you have two [idle_timeout] sections in your printer file, that may explain your status error

josurr, 2023-10-19:
Thank you for that insight!

I did have another question, is there an example "variables.cfg" file that can be used as a reference for creating one that is specific to my printer?

Perrotn, 2023-10-19:
My understanding is that variables.cfg is automatically created and is used to store variables like the z offset of each surfaces. Mine look like that:
[Variables]
bed_surfaces = {'active': 'glass', 'available': {'pei_black': {'offset': 0.0}, 'pei_gold': {'offset': -0.4}, 'glass': {'offset': -0.4}}, 'endstop_z': 0.0, 'probe_z': 1.49}

closed 2023-11-07 by jschuh